YosysHQ/nextpnr #1248: Oassing constraints to nextpnr-himbaechel

janschiefer, 2023-11-18:
How do I pass physical constraints / pin mappings to nextpnr-himbaechel, for example with Gowin chips ( .cst, .pcf )?

yrabbit, 2023-11-18:
Well, just like the examples coming from apicula (cst, I don't know anything about pcf):

https://github.com/YosysHQ/apicula/blob/a4f04e0c481d0ab5feb828a311cca1a76c7b18b6/examples/himbaechel/Makefile.himbaechel#L147

janschiefer, 2023-11-18:
Nice!

However, nextpnr-himbaechel  now finishes, but fails to find a clock in the design, while nextpnr-gowin synthesizes finishes without a problem ( an recognized the clock ).

led.v:
```
module led (
    input clk,          // clk input
    input rst_i,        // reset input
    output reg [5:0] led    // 6 LEDS pin
);

reg [23:0] counter;

always @(posedge clk or negedge rst_i) begin
    if (!rst_i)
        counter <= 24'd0;
    else if (counter < 24'd1349_9999)       // 0.5s delay
        counter <= counter + 1'd1;
    else
        counter <= 24'd0;
end

always @(posedge clk or negedge rst_i) begin
    if (!rst_i)
        led <= 6'b101110;
    else if (counter == 24'd1349_9999)       // 0.5s delay
        led[5:0] <= {led[4:0],led[5]};
    else
        led <= led;
end

endmodule
```

led.cst:
```
//Part Number: GW1NR-LV9QN88PC6/I5

IO_LOC "clk" 52;
IO_LOC "led[0]" 10;
IO_LOC "led[1]" 11;
IO_LOC "led[2]" 13;
IO_LOC "led[3]" 14;
IO_LOC "led[4]" 15;
IO_LOC "led[5]" 16;
IO_LOC "rst_i" 4;
```

build-gowin.sh:
```
#!/bin/bash
rm *.json
rm *.fs
yosys -D LEDS_NR=6 -D OSC_TYPE_OSC -p "read_verilog led.v; synth_gowin -json synth.json -top led" 
nextpnr-gowin --json synth.json --write out.json --device GW1NR-LV9QN88PC6/I5 --cst led.cst
gowin_pack -d GW1N-9C -o out.fs out.json
```

build-himbaechel.sh:
```
#!/bin/bash
rm *.json
rm *.fs
yosys -D LEDS_NR=6 -D OSC_TYPE_OSC -p "read_verilog led.v; synth_gowin -json synth.json -top led" 
nextpnr-himbaechel --json synth.json --write out.json --device GW1NR-LV9QN88PC6/I5 --vopt family=GW1N-9C --vopt cst=led.cst 
gowin_pack -d GW1N-9C -o out.fs out.json
```

nextpnr-gowin:
[... 354 lines cut ...]
Info: Running simulated annealing placer for refinement.
Info:   at iteration #1: temp = 0.000000, timing cost = 0, wirelen = 340
Info:   at iteration #5: temp = 0.000000, timing cost = 0, wirelen = 314
Info:   at iteration #5: temp = 0.000000, timing cost = 0, wirelen = 314 
Info: SA placement time 0.01s

Info: No Fmax available; no interior timing paths found in design.
Info: Checksum: 0x818f8229
Info: Routing globals...
Info:     routed net 'clk_IBUF_I_O' using global resources

Info: Routing..
Info: Setting up routing queue.
Info: Routing 385 arcs.
Info:            |   (re-)routed arcs  |   delta    | remaining|       time spent     |
Info:    IterCnt |  w/ripup   wo/ripup |  w/r  wo/r |      arcs| batch(sec) total(sec)|
Info:        391 |        4        334 |    4   334 |         0|       1.48       1.48|
Info: Routing complete.
Info: Router1 time 1.48s
Info: Checksum: 0xc1d2c083

**Info: No Fmax available; no interior timing paths found in design.**

Info: Program finished normally.
```

yrabbit, 2023-11-18:
And the problem is...?
Interesting blinking, btw.

https://github.com/YosysHQ/nextpnr/assets/6075465/f96c36c8-1269-40e5-ac60-9864d64bd80b

janschiefer, 2023-11-19:
Ok. So just the determination of fmax is not implemented yet.
 
Thank you!

*blink* *blink*

https://github.com/YosysHQ/nextpnr/assets/29553827/e8fd0eb2-ee17-4f54-9b4e-f6dadbefe988

closed 2023-11-19 by janschiefer