YosysHQ/nextpnr #1254: No option named X for enum EBR3.PDPSC16K_MODE.DATA_WIDTH_R in tile CIB_R29C11:EBR_9

slagernate, 2023-11-29:
Using Amaranth 0.4 (w/ Yosys 0.35) with prjoxide 0.1 and nextpnr 0.6-152 (these are all latest AFAIK) but getting the following:

```Info: Program finished normally.
thread 'main' panicked at 'No option named X for enum EBR3.PDPSC16K_MODE.DATA_WIDTH_R in tile CIB_R29C11:EBR_9.
Valid options are: X1, X18, X2, X32, X36, X4, X9
Please make sure Oxide and nextpnr are up to date and input source code is meaningful. If they are, consider reporting this as an issue.', prjoxide/src/chip.rs:415:75```

While a manual fix of the 'X' option in Amaranth's generated Verilog allows for a successful build, the EBR does not appear to behave normally.

Can make an MRE at some point.

slagernate, 2023-11-29:
Here is the EBR instantiation from yosys.debug.v:

```  
PDPSC16K #(
    .ASYNC_RST_RELEASE("ASYNC"),
    .CSDECODE_R("111"),
    .CSDECODE_W("111"),
    .DATA_WIDTH_R("X         4"),
    .DATA_WIDTH_W("X         4"),
    .ECC("DISABLED"),
    .GSR("DISABLED"),
    .INITVAL_00("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_01("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_02("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_03("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_04("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_05("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_06("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_07("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_08("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_09("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_0A("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_0B("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_0C("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_0D("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_0E("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_0F("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_10("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_11("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_12("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_13("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_14("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_15("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_16("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_17("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_18("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_19("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_1A("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_1B("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_1C("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_1D("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_1E("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_1F("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_20("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_21("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_22("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_23("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_24("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_25("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_26("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_27("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_28("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_29("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_2A("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_2B("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_2C("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_2D("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_2E("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_2F("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_30("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
    .INITVAL_31("0x00000000000000000000000000000000000000000000000000000000000000000000000000000000"),
[... cut ...] $$1.en$8 , \U$$0.U$$1.en$8 , \U$$0.U$$0.U$$0_w_data [3:0] }),
    .DO({ \U$$0.U$$0.U$$0.storage.0.0_DO [35:4], \U$$0.U$$0.U$$0.storage_r_data [3:0] }),
    .RST(\U$$0.U$$1.en$8 )
  );
```

The error originates from `    .DATA_WIDTH_R("X         4"),` where a frew spaces are inserted (to recap, this can compile if the spaces are manually removed, yet the behaviour appears to be incorrect, reading out zeros all or if not most of the time). 

Making an MRE for the Nexus Crosslink-NX eval board now.

slagernate, 2023-11-30:
And here is example amaranth which reproduces the panic above (albeit the memory behaves as expected on the crosslink-nx eval board):
Note: using a [Lattice Vendor platform](https://github.com/slagernate/amaranth/blob/main/amaranth/vendor/lattice_nexus.py) and a [eval board definition file](https://github.com/slagernate/amaranth-boards/blob/main/amaranth_boards/lifcl_evn.py)
```
import itertools
import os
from amaranth import *
from amaranth.lib.fifo import *
from amaranth.build import *
​
​
from amaranth_boards.boards.lifcl_evn import LIFCLEVNPlatform
​
class DebugModule(Elaboratable):
​
    def __init__(self):
​
        self.fifo = SyncFIFOBuffered(width=32, depth=4096)
        self.data_o = Signal(12)
​
    def elaborate(self, platform):
        m = Module()
        m.submodules += self.fifo
​
        clk_div = Signal(24)
        counter = Signal(32)
        buf = Signal(32)
​
        m.d.sync += clk_div.eq(clk_div + 1)
​
        with m.If(clk_div == 0 & self.fifo.r_rdy):
            m.d.sync += self.fifo.r_en.eq(1)
​
        with m.If(self.fifo.r_en):
            m.d.sync += buf.eq(self.fifo.r_data)
            m.d.sync += self.fifo.r_en.eq(0)
​
        with m.If(self.fifo.w_rdy):
            m.d.sync += self.fifo.w_data.eq(counter)
            m.d.sync += self.fifo.w_en.eq(1)
            m.d.sync += counter.eq(counter + 1)
        with m.Else():
            m.d.sync += self.fifo.w_en.eq(0)
​
        for i in range(8):
            m.d.comb += self.data_o[i].eq(buf[4*i:(4*i)+4].any())
​
        def get_all_resources(name):
            resources = []
            for number in itertools.count():
                try:
                    resources.append(platform.request(name, number))
                except ResourceError:
                    break
            return resources
        
        leds = [res.o for res in get_all_resources("led")]
​
        m.d.comb += Cat(leds).eq(self.data_o)
        print(len(leds))
​
        return m
    
​
if __name__ == "__main__":
    os.environ["AMARANTH_synth_opts"] = "-vm yosys.debug.v"
​
    platform = LIFCLEVNPlatform(toolchain="Oxide")
    platform.build(DebugModule(), do_program=False)
 ```

gatecat, 2023-12-01:
See https://github.com/YosysHQ/yosys/pull/4048  - I haven't yet tested it on hardware, but at least this fixes the build failure.

closed 2023-12-03 by gatecat

slagernate, 2023-12-05:
No issues in hardware on my side.