YosysHQ/nextpnr #1261: Nexus:  Cell type 'CCU2'...not supported by this device.  Encountered when using 'synth_nexus -abc9' in Yosys.

tcal-x, 2023-12-18:
This may or may not be a nextpnr bug, but I'm starting here to hopefully get a pointer to the real issue.

In a current LiteX setup, if you `cd` to `litex-boards/litex-boards/targets/` and run
```
$ ./lattice_crosslink_nx_evn.py --build --toolchain=oxide --yosys-abc9
```
it will eventually fail in nextpnr-nexus with 
```
ERROR: Cell type 'CCU2' instantiated as 'main_crg_por_count_FD1P3JX_Q_D_CCU2_S1_CIN_CCU2_COUT' is not supported by this device.
```

If you don't include the `--yosys-abc9`, it succeeds.   

I'll attach a zip with the Yosys output json from each case, the pdc file, and the command line to rerun `nextpnr-nexus`: [demo.zip](https://github.com/YosysHQ/nextpnr/files/13700680/demo.zip)

One diff I noticed between the json files is this, *which might be completely irrelevant*:
```
// without -abc9 (the version that works):
    "CCU2": {
      "attributes": {
        "abc9_box": "00000000000000000000000000000001",
        "blackbox": "00000000000000000000000000000001",
        "src": "/usr/local/bin/../share/yosys/nexus/cells_sim.v:197.1-243.10"
      },
```
```
// with -abc9 (the version that results in the error):
    "CCU2": {
      "attributes": {
        "blackbox": "00000000000000000000000000000001",
        "src": "/usr/local/bin/../share/yosys/nexus/cells_sim.v:197.1-243.10"
      },
```

Versions used:
```
"nextpnr-nexus" -- Next Generation Place and Route (Version nextpnr-0.6-154-g535709a9)
Yosys 0.36+42 (git sha1 70d35314d, clang 14.0.0-1ubuntu1.1 -fvisibility-inlines-hidden -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -fPIC -Os)
```

gatecat, 2023-12-18:
Thanks for reporting!

I've improved the bad error message, occurring because of a failure in carry chain packing, in https://github.com/YosysHQ/nextpnr/pull/1262

However, this doesn't fix the root cause, which seems to be that ABC9 is now converting the correct missing carry input on the start of a carry chain, to a carry input connected to zero (which can't necessarily be packed). That would require a Yosys MCVE and fix, which I will look at over the next few days if no-one else gets there first.

tcal-x, 2023-12-19:
Thanks @gatecat for the quick response.   I think I have a pretty small reproducer.

gatecat, 2024-09-26:
The nextpnr issue is fixed here so I'll close this ticket.