YosysHQ/nextpnr #1272: NextPNR-nexus reports PCLKDIV primitive not supported on LICL-40

polymerizedsage, 2024-01-09:
When trying to synthesize a design for the LICL-40, NextPNR-nexus reports `Cell type 'PCLKDIV' instantiated as 'U$$5' is not supported by this device.` despite the primitive being listed on the Lattice/Radiant documentation and working properly when building using the Radiant toolchain. 

The primitive is instantiated in Amaranth HDL as follows:
```        
m.submodules += Instance("PCLKDIV", 
                                 p_DIV_PCLKDIV      = "X2",
                                 i_CLKIN            = clk48.i,
                                 i_LSRPDIV          = Const(0),
                                 o_CLKOUT           = clk_ser,
                                 )
```

Using yowasp-nextpnr-nexus 0.6.0.0.post445

slagernate, 2024-04-11:
Appears `IDDRX4` is also not yet in prjxode, despite it working with the Lattice Radiant tool chain:
```
        def get_iddrx4(sclk, eclk, d, q0, q1, q2, q3, q4, q5, q6, q7):
            for bit in range(len(d)):
                m.submodules += Instance("IDDRX4",
                    i_SCLK=sclk,
                    i_ECLK=eclk,
                    i_RST=Const(0),
                    i_ALIGNWD=Const(0),
                    i_D=d[bit],
                    o_Q0=q0[bit],
                    o_Q1=q1[bit],
                    o_Q2=q2[bit],
                    o_Q3=q3[bit],
                    o_Q4=q4[bit],
                    o_Q5=q5[bit],
                    o_Q6=q6[bit],
                    o_Q7=q7[bit],
                )
```
results in:
```
ERROR: Cell type 'IDDRX4' instantiated as 'pin_xxx_0.U$1' is not supported by this device.
```

whitequark, 2024-04-11:
@slagernate If this is for Amaranth, please note that we are (temporarily) dropping all gearing ratios higher than 1:2 (DDR) for the 0.5 release.