YosysHQ/nextpnr #1276: Cut a release with Gowin Himbaechel uarch

pepijndevos, 2024-01-20:
Can we please have a nextpnr release that supports the Gowin himbaechel backend? It's been half a year since the last release.

We keep getting issues from people asking for devices that are already supported but not in any release version of nextpnr. 
We also carry a huge amount of legacy code to support nextpnr-gowin which is increasingly lagging behind, with no support for BRAM and the entire GW2A family of devices.

Currently the process for building nxtpnr-himbaechel with the Gowin uarch is uncodumented in either Nextpnr (which only mentions nextpnr-gowin) or Apicula (which just points to Nextpnr) so unless this changes soon we'll have to go back to telling people to install the latest git master.

gatecat, 2024-01-23:
Sounds good, I have this roughly planned for tomorrow unless anyone objects...

gatecat, 2024-01-24:
[Done!](https://github.com/YosysHQ/nextpnr/releases/tag/nextpnr-0.7)

closed 2024-01-24 by gatecat