YosysHQ/nextpnr #1288: Installation on Ubuntu 22.04

Ownasaurus, 2024-02-03:
Hello! Thank you for your time in advance. I last used nextpnr-ecp5 about 3 years ago and wanted to update the whole toolchain and resume development.

- Latest master Libtrellis seemed to install no problem to /usr/local
- Latest master Yosys seemed to install no problem

But I'm having issues with nextpnr! From the root, I'm running `cmake . -DARCH=ecp5 -DTRELLIS_INSTALL_PREFIX=/usr/local`
It detects boost:
```
-- Found Boost: /usr/include (found version "1.74.0") found components: filesystem program_options iostreams system thread regex chrono date_time atomic 
-- Found Boost: /usr/include (found version "1.74.0") found components: program_options filesystem system
```

But then when I try to make, I get the following error:
`ImportError: libboost_thread.so.1.71.0: cannot open shared object file: No such file or directory`

Why is it looking for 1.71 if version 1.74 was found earlier? I am sorry if I am simply being a newbie, and I am grateful for any assistance. Thanks again!

gatecat, 2024-02-03:
This is because it's trying to load a version of libtrellis that was built against an older version of boost.

Make sure nextpnr is pointed to the right trellis path, and libtrellis is cleanly built and reinstalled.

Ownasaurus, 2024-02-03:
Thank you so much for your reply! I deleted my local git **prjtrellis** repo, re-cloned, and followed the instructions. nextpnr gave the same error on compiling though :(

So, I deleted my local git **nextpnr** repo, re-cloned, and followed the instructions. Now it is successfully building!

I wish I knew what I was doing wrong and why it was looking for an old version, but alas, all is well now. Is it possibly because some of the old cmake settings were saved? I'm not very experienced with cmake. If I pulled the latest version, but I had some files left over from 3+ years ago, could that have interfered? Is one expected to do a 'make clean' equivalent with cmake?

Thanks again!

gatecat, 2024-02-03:
Yeah, my guess is there was an older install of trellis installed at a different path (say `/usr` instead of `/use/local`) that CMake had cached the path to.

Ownasaurus, 2024-02-03:
Thanks again!

closed 2024-02-03 by Ownasaurus