YosysHQ/nextpnr #1292: Make fail

yathAg, 2024-02-18:
I bused the following cmake command `cmake -DARCH=ice40 -DCMAKE_INSTALL_PREFIX=/usr/local .`

which resulted in 

```
-- The CXX compiler identification is GNU 9.4.0
-- The C compiler identification is GNU 9.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Building with IPO
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Performing Test HAS_Wno-unused-parameter
-- Performing Test HAS_Wno-unused-parameter - Success
-- Performing Test HAS_Wno-missing-field-initializers
-- Performing Test HAS_Wno-missing-field-initializers - Success
-- Performing Test HAS_Wno-array-bounds
-- Performing Test HAS_Wno-array-bounds - Success
-- Performing Test HAS_Wno-format-truncation
-- Performing Test HAS_Wno-format-truncation - Success
-- Found Python3: /usr/bin/python3.9 (found suitable version "3.9.18", minimum required is "3.5") found components: Interpreter 
-- Found Python3: /usr/include/python3.8 (found suitable version "3.8.10", minimum required is "3.5") found components: Development Development.Module Development.Embed 
-- Found Boost: /usr/local/include (found version "1.80.0") found components: filesystem program_options iostreams system thread regex chrono atomic 
-- Found Boost: /usr/local/include (found version "1.80.0") found components: program_options filesystem system 
-- Configuring architecture: ice40
-- Enabled iCE40 devices: 384;1k;5k;u4k;8k
-- Found Python3: /usr/bin/python3.9 (found suitable version "3.9.18", minimum required is "3.5") found components: Interpreter 
-- IceStorm install prefix: /usr/local
-- icebox data directory: /usr/local/share/icebox
-- Using iCE40 chipdb: /home/yatharth/project_vsdsquadronFM/nextpnr/ice40/chipdb
-- Configuring done (1.3s)
-- Generating done (0.0s)
-- Build files have been written to: /home/yatharth/project_vsdsquadronFM/nextpnr
```

followed by `make -j$(nproc)`

which throws the error

```
[  1%] Generating chipdb/chipdb-384.bba
[  2%] Building CXX object bba/CMakeFiles/bbasm.dir/main.cc.o
[  4%] Generating chipdb/chipdb-1k.bba
[  5%] Linking CXX executable bbasm
/usr/bin/ld: CMakeFiles/bbasm.dir/main.cc.o: in function `boost::filesystem::path::stem() const':
/usr/local/include/boost/filesystem/path.hpp:845: undefined reference to `boost::filesystem::path::stem_v3() const'
collect2: error: ld returned 1 exit status
make[2]: *** [bba/CMakeFiles/bbasm.dir/build.make:100: bba/bbasm] Error 1
make[1]: *** [CMakeFiles/Makefile2:229: bba/CMakeFiles/bbasm.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  7%] Generating chipdb/chipdb-5k.bba
[  8%] Generating chipdb/chipdb-u4k.bba
[  9%] Generating chipdb/chipdb-8k.bba
[  9%] Built target chipdb-ice40-bbas
make: *** [Makefile:136: all] Error 2
```

Any idea what the issue cloud be? Thanks for the help!

rowanG077, 2024-02-18:
This can happen if boost libraries are not linked in properly. Can you attach the output of `make VERBOSE=1`?

yathAg, 2024-02-18:
```
cd /home/yatharth/project_vsdsquadronFM/nextpnr/bba && /home/yatharth/.local/lib/python3.9/site-packages/cmake/data/bin/cmake -E cmake_link_script CMakeFiles/bbasm.dir/link.txt --verbose=1
/usr/bin/c++ -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -Wno-array-bounds -Wno-format-truncation -fPIC -O3 -g -pipe CMakeFiles/bbasm.dir/main.cc.o -o bbasm  /usr/lib/x86_64-linux-gnu/libboost_program_options.so /usr/lib/x86_64-linux-gnu/libboost_filesystem.so /usr/lib/x86_64-linux-gnu/libboost_system.so 
/usr/bin/ld: CMakeFiles/bbasm.dir/main.cc.o: in function `boost::filesystem::path::stem() const':
/usr/local/include/boost/filesystem/path.hpp:845: undefined reference to `boost::filesystem::path::stem_v3() const'
collect2: error: ld returned 1 exit status
make[2]: *** [bba/CMakeFiles/bbasm.dir/build.make:100: bba/bbasm] Error 1
make[2]: Leaving directory '/home/yatharth/project_vsdsquadronFM/nextpnr'
make[1]: *** [CMakeFiles/Makefile2:229: bba/CMakeFiles/bbasm.dir/all] Error 2
make[1]: Leaving directory '/home/yatharth/project_vsdsquadronFM/nextpnr'
make: *** [Makefile:136: all] Error 2
```

yathAg, 2024-02-18:
Yep, it was an issue with my boost version and the way the path was linked. I built boost from source and updated the CMake cache. It works now!

closed 2024-02-18 by yathAg