YosysHQ/nextpnr #1297: undefined symbol: _ZNK5boost10filesystem4path7stem_v3Ev

gareth-anthony-hulse, 2024-03-02:
NextPNR fails to find a Boost symbol with iCE40 and ECP5. Don't know about the others. I am using [nextpnr-git 0.6.r100.g8e84006e-1](https://aur.archlinux.org/packages/nextpnr-git). 

ECP5:
```
nextpnr-ecp5: symbol lookup error: nextpnr-ecp5: undefined symbol: _ZNK5boost10filesystem4path7stem_v3Ev
make: *** [Makefile:10: blinky_out.config] Error 127
```
iCE40:
```
nextpnr-ice40: symbol lookup error: nextpnr-ice40: undefined symbol: _ZNK5boost10filesystem4path7stem_v3Ev
make: *** [Makefile:11: /home/gareth-anthony-hulse/Projects/gateware/lattice-ice40hx-8k-example/build/lattice-ice40hx-8k-example.asc] Error 127
```
I am using Antergos Linux, but the repos have been changed to Arch Linux.
**Specs:** https://linux-hardware.org/?probe=92b98e9789

rowanG077, 2024-03-02:
Probably the same as: https://github.com/YosysHQ/nextpnr/issues/1292

gareth-anthony-hulse, 2024-03-03:
@rowanG077 I tried building and installing Boost from [source](https://github.com/boostorg/boost), but still got the same error messages.
output of `gcc -xc -E -v -`:
```
Using built-in specs.
COLLECT_GCC=/usr/bin/gcc
Target: x86_64-pc-linux-gnu
Configured with: /build/gcc/src/gcc/configure --enable-languages=ada,c,c++,d,fortran,go,lto,m2,objc,obj-c++ --enable-bootstrap --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=https://bugs.archlinux.org/ --with-build-config=bootstrap-lto --with-linker-hash-style=gnu --with-system-zlib --enable-__cxa_atexit --enable-cet=auto --enable-checking=release --enable-clocale=gnu --enable-default-pie --enable-default-ssp --enable-gnu-indirect-function --enable-gnu-unique-object --enable-libstdcxx-backtrace --enable-link-serialization=1 --enable-linker-build-id --enable-lto --enable-multilib --enable-plugin --enable-shared --enable-threads=posix --disable-libssp --disable-libstdcxx-pch --disable-werror
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 13.2.1 20230801 (GCC) 
COLLECT_GCC_OPTIONS='-E' '-v' '-mtune=generic' '-march=x86-64'
 /usr/lib/gcc/x86_64-pc-linux-gnu/13.2.1/cc1 -E -quiet -v - -mtune=generic -march=x86-64 -dumpbase -
ignoring nonexistent directory "/usr/lib/gcc/x86_64-pc-linux-gnu/13.2.1/../../../../x86_64-pc-linux-gnu/include"
#include "..." search starts here:
#include <...> search starts here:
 /usr/lib/gcc/x86_64-pc-linux-gnu/13.2.1/include
 /usr/local/include
 /usr/lib/gcc/x86_64-pc-linux-gnu/13.2.1/include-fixed
 /usr/include
End of search list.
```
Output of `ldd /usr/bin/nextpnr-ice40`:
```
	linux-vdso.so.1 (0x00007fff0f0c2000)
	libboost_filesystem.so.1.85.0 => /usr/local/lib/libboost_filesystem.so.1.85.0 (0x00007dce37719000)
	libboost_program_options.so.1.85.0 => /usr/local/lib/libboost_program_options.so.1.85.0 (0x00007dce376ae000)
	libboost_thread.so.1.85.0 => /usr/local/lib/libboost_thread.so.1.85.0 (0x00007dce37690000)
	libpython3.11.so.1.0 => /usr/lib/libpython3.11.so.1.0 (0x00007dce29000000)
	libQt5Widgets.so.5 => /usr/lib/libQt5Widgets.so.5 (0x00007dce28800000)
	libQt5Gui.so.5 => /usr/lib/libQt5Gui.so.5 (0x00007dce28000000)
	libQt5Core.so.5 => /usr/lib/libQt5Core.so.5 (0x00007dce27a00000)
	libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x00007dce27600000)
	libm.so.6 => /usr/lib/libm.so.6 (0x00007dce28f14000)
	libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x00007dce37669000)
	libc.so.6 => /usr/lib/libc.so.6 (0x00007dce2741e000)
	libboost_atomic.so.1.85.0 => /usr/local/lib/libboost_atomic.so.1.85.0 (0x00007dce3765f000)
	libGL.so.1 => /usr/lib/libGL.so.1 (0x00007dce2877a000)
	libpng16.so.16 => /usr/lib/libpng16.so.16 (0x00007dce37623000)
	libz.so.1 => /usr/lib/libz.so.1 (0x00007dce295e6000)
	libharfbuzz.so.0 => /usr/lib/libharfbuzz.so.0 (0x00007dce278f2000)
	libmd4c.so.0 => /usr/lib/libmd4c.so.0 (0x00007dce28efe000)
	libsystemd.so.0 => /usr/lib/libsystemd.so.0 (0x00007dce2868d000)
	libdouble-conversion.so.3 => /usr/lib/libdouble-conversion.so.3 (0x00007dce28ee8000)
	libicui18n.so.74 => /usr/lib/libicui18n.so.74 (0x00007dce27000000)
	libicuuc.so.74 => /usr/lib/libicuuc.so.74 (0x00007dce26c00000)
	libpcre2-16.so.0 => /usr/lib/libpcre2-16.so.0 (0x00007dce27f72000)
	libzstd.so.1 => /usr/lib/libzstd.so.1 (0x00007dce2734b000)
	libglib-2.0.so.0 => /usr/lib/libglib-2.0.so.0 (0x00007dce26eb4000)
	/lib64/ld-linux-x86-64.so.2 => /usr/lib64/ld-linux-x86-64.so.2 (0x00007dce377a8000)
	libGLdispatch.so.0 => /usr/lib/libGLdispatch.so.0 (0x00007dce26b48000)
	libGLX.so.0 => /usr/lib/libGLX.so.0 (0x00007dce28eb6000)
	libfreetype.so.6 => /usr/lib/libfreetype.so.6 (0x00007dce26a7a000)
	libgraphite2.so.3 => /usr/lib/libgraphite2.so.3 (0x00007dce27f4f000)
	libcap.so.2 => /usr/lib/libcap.so.2 (0x00007dce295da000)
	libgcrypt.so.20 => /usr/lib/libgcrypt.so.20 (0x00007dce26931000)
	liblz4.so.1 => /usr/lib/liblz4.so.1 (0x00007dce278cd000)
	liblzma.so.5 => /usr/lib/liblzma.so.5 (0x00007dce2789a000)
	libicudata.so.74 => /usr/lib/libicudata.so.74 (0x00007dce24a00000)
	libpcre2-8.so.0 => /usr/lib/libpcre2-8.so.0 (0x00007dce26e19000)
	libX11.so.6 => /usr/lib/libX11.so.6 (0x00007dce267ee000)
	libbz2.so.1.0 => /usr/lib/libbz2.so.1.0 (0x00007dce28ea3000)
	libbrotlidec.so.1 => /usr/lib/libbrotlidec.so.1 (0x00007dce2867e000)
	libgpg-error.so.0 => /usr/lib/libgpg-error.so.0 (0x00007dce267c8000)
	libxcb.so.1 => /usr/lib/libxcb.so.1 (0x00007dce2679d000)
	libbrotlicommon.so.1 => /usr/lib/libbrotlicommon.so.1 (0x00007dce2677a000)
	libXau.so.6 => /usr/lib/libXau.so.6 (0x00007dce295d5000)
	libXdmcp.so.6 => /usr/lib/libXdmcp.so.6 (0x00007dce28e9b000)
```
Output of `ldd /usr/bin/nextpnr-ecp5`:
```
[... 66 lines cut ...]
bimap                    crc.hpp                      gil                           logic                          pfr.hpp                       safe_numerics                  type_traits.hpp
bimap.hpp                cregex.hpp                   gil.hpp                       make_default.hpp               phoenix                       scope                          units
bind                     cstdfloat.hpp                graph                         make_shared.hpp                phoenix.hpp                   scoped_array.hpp               unordered
bind.hpp                 cstdint.hpp                  hana                          make_unique.hpp                pointee.hpp                   scoped_ptr.hpp                 unordered_map.hpp
[... cut ...]
compat                   exception_ptr.hpp            io_fwd.hpp                    multi_index_container.hpp      property_tree                 static_string.hpp              xpressive
compatibility            fiber                        iostreams                     multiprecision                 proto                         stl_interfaces                 yap
compressed_pair.hpp      filesystem                   is_placeholder.hpp            mysql                          ptr_container                 swap.hpp
```

gareth-anthony-hulse, 2024-03-03:
It just came to mind that I might have an old version of NextPNR installed. tried `which nextpnr-ecp5` and sure enough the output was `/usr/local/bin/nextpnr-ecp5`. I completely forgot that I had built and installed a source version of NextPNR many months ago, up until now. Removed NextPNR for ECP5 and iCE40 from `/usr/local/bin/`. And everything now works.

closed 2024-03-03 by gareth-anthony-hulse