YosysHQ/nextpnr #1304: nextpnr-ecp5 build failed during chipdb generation

solverstin, 2024-03-27:
I've attempt to build nextpnr from current github repository, and it fails to build ECP5 target. My setup is Debian-12, yosys installed from debian package and prjtrellis installed from its current github repository. The result is:
`~/work/fpga/ecp/nextpnr$ cmake . -DARCH=ecp5
-- Building with IPO
-- Found Python3: /usr/include/python3.11 (found suitable version "3.11.2", minimum required is "3.5") found components: Development Development.Module Development.Embed 
-- Found Boost: /usr/include (found version "1.74.0") found components: filesystem program_options iostreams system thread regex chrono date_time atomic 
-- Found Boost: /usr/include (found version "1.74.0") found components: program_options filesystem system 
-- Configuring architecture: ecp5
-- Enabled ECP5 devices: 25k;45k;85k
-- Found Python3: /usr/bin/python3 (found suitable version "3.11.2", minimum required is "3.5") found components: Interpreter 
-- Trellis install prefix: /usr/local
-- Trellis library directory: /usr/local/lib/trellis
-- Trellis data directory: /usr/local/share/trellis
-- Using ECP5 chipdb: ecp5/chipdb
-- Configuring done
-- Generating done
-- Build files have been written to: .
~/work/fpga/ecp/nextpnr$ make
[  1%] Building CXX object bba/CMakeFiles/bbasm.dir/main.cc.o
[  3%] Linking CXX executable bbasm
[  3%] Built target bbasm
[  4%] Generating chipdb/chipdb-25k.bba
Traceback (most recent call last):
  File "ecp5/trellis_import.py", line 709, in <module>
    main()
  File "ecp5/trellis_import.py", line 692, in main
    chip = pytrellis.Chip(dev_names[args.device])
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
RuntimeError: No such node (row_bias)
make[2]: *** [ecp5/CMakeFiles/chipdb-ecp5-bbas.dir/build.make:77: ecp5/chipdb/chipdb-25k.bba] Błąd 1
make[1]: *** [CMakeFiles/Makefile2:255: ecp5/CMakeFiles/chipdb-ecp5-bbas.dir/all] Błąd 2
make: *** [Makefile:136: all] Błąd 2
`

Other tarets (ice40, generic and himbaechel) build correctly.

There must be something obvious I did wrong, but I cannot figure it out. Any help appreciated.

mmicko, 2024-03-27:
Hi. Please do check your local checkout of `prjtrellis` and make sure that `database` submodule is also properly updated.  My assumption is that issue could be there. Also after any change make sure to go into `libtrellis` and execute `make install` frin there.

solverstin, 2024-03-27:
OK, better, but failed nonetheless a little bit later.

> $ make
[  3%] Built target bbasm
[  4%] Generating chipdb/chipdb-25k.bba
Traceback (most recent call last):
  File "ecp5/trellis_import.py", line 749, in <module>
    main()
  File "ecp5/trellis_import.py", line 742, in main
    process_devices_db(chip.info.family, [ chip.info.name, chip.info.name.replace("LFE5UM","LFE5U") ])
  File "ecp5/trellis_import.py", line 260, in process_devices_db
    for varname, vardata in sorted(devicedb["families"][family]["devices"][dev]["variants"].items()):
                                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
KeyError: 'variants'
make[2]: *** [ecp5/CMakeFiles/chipdb-ecp5-bbas.dir/build.make:77: ecp5/chipdb/chipdb-25k.bba] Błąd 1
make[1]: *** [CMakeFiles/Makefile2:255: ecp5/CMakeFiles/chipdb-ecp5-bbas.dir/all] Błąd 2
make: *** [Makefile:136: all] Błąd 2

mmicko, 2024-03-27:
Just tested with latest and got now issues. 
Log points that your database.json is not right.
Please go to your local `prjtrellis/database` and run `git log -n 1` from there, it should point to version `4dda149b9e4f1753ebc8b011ece2fe794be1281a` if it is not that version make sure to update submodule.

solverstin, 2024-03-27:
I have the same version, here:

> $ git log -n 1
commit 4dda149b9e4f1753ebc8b011ece2fe794be1281a (HEAD, origin/master, origin/HEAD, master)
Author: Miodrag Milanovic <mmicko@gmail.com>
Date:   Fri Sep 22 11:20:25 2023 +0200

    Fuzzed data for rest of SLICE connections

I also did the following "update" in the database submodule:

> $ git pull
Nie jesteś obecnie w gałęzi.
Podaj, z jaką gałęzią scalić.
Więcej szczegółów w git-pull(1).
>
>    git pull <zdalne-repozytorium> <gałąź>
>
> $ git branch
>\* (HEAD odłączone na 4dda149)
  master
>$ git checkout master
Przełączono na gałąź „master”
Twoja gałąź jest na bieżąco z „origin/master”.
>$ git branch 
>\* master
>$ git branch -a
>\* master
  remotes/origin/HEAD -> origin/master
  remotes/origin/efb_support
  remotes/origin/gh-pages
  remotes/origin/master
  remotes/origin/micko/ecp5_update

But nothing needed any pdate, when I did another "make install" in the libtrellis subdir. The problem must be somwhere else...

mmicko, 2024-03-27:
Default location for install, is in `/usr/local` so file at `/usr/local/share/trellis/database/devices.json` should match device.json from database sub directory. 
Just to be sure you are using Trellis from proper location, run `cat CMakeCache.txt | grep TRELLIS` it should give you:

```
TRELLIS_INSTALL_PREFIX:STRING=/usr/local
TRELLIS_LIBDIR:PATH=/usr/local/lib/trellis
TRELLIS_PROGRAM_PREFIX:STRING=
```

solverstin, 2024-03-27:
I get exactly the same values.

But I've looked into the `devices.json`, and not all the defices do have a `variant` keyword within it. Could that be the reason `*import.py` script fails here: 
>  File "ecp5/trellis_import.py", line 260, in process_devices_db
>    for varname, vardata in sorted(devicedb["families"][family]["devices"][dev]["variants"].items()):
>KeyError: 'variants'

mmicko, 2024-03-27:
Ah, it looks like you are actually on branch, or have applied some of commits from `micko/ecp5_update` branch. On current master there is no use of "variants" for ECP5 https://github.com/YosysHQ/nextpnr/blob/master/ecp5/trellis_import.py#L260

solverstin, 2024-03-27:
Oh, this is it!

I've switched to it in a desperate search for reasons why my build failed.... and the actual reason was the missing sobmodule in my prjtrellis build.

Thenx for your help!

BTW: would you think, that there should be some guard in the `prjtrellis` for such misshaps? Like `make install` checking-if/depensing--on the `devices.json` file, and doing apropriate `git submodule init` if its missing?

mmicko, 2024-03-27:
Glad it is solved. Not sure if this would actually help, doing such checks would complicate maintaining and still there will be room for user error.

closed 2024-03-27 by mmicko