YosysHQ/nextpnr #1305: On iCE40, use SB_IO PIN_TYPE to avoid unnecessary placement constraint on buffers

whitequark, 2024-03-27:
Right now it appears that regardless of PIN_TYPE, the CLOCK_ENABLE and {INPUT,OUTPUT}_CLOCK in a pin pair with z=0,1 must match. This is mostly fine for clocks but gets *extremely* annoying for CLOCK_ENABLE, and worse, the constraint still has to be preserved if you use the pin as a comb input/output.

Nextpnr could take this into account and ignore the nets tied to CLOCK_ENABLE/*_CLOCK if the PIN_TYPE instructs it so.

gatecat, 2024-05-03:
I believe this should be implemented via [this function](https://github.com/YosysHQ/nextpnr/blob/3f2451f8d78c88e299cb421539add41eadae17bf/ice40/arch_place.cc#L80-L83) already - can you provide an example where this isn't working?

whitequark, 2024-05-03:
@wanda-phi Do you recall where we hit this?

whitequark, 2024-05-03:
Oh it could have been!