YosysHQ/nextpnr #1306: Symbol not found: __ZNK5boost10filesystem4path7stem_v3Ev

zaun, 2024-03-30:
Not sure if this is the best place to ask or not, but I can compile and install nextpnr but when I try to run the app it gives me a Symbol not found error. I'm on MacOS 14.3.1 (23D60)

**Config:**

```
╭─justinzaun@Hogsmeade ~/Development/nextpnr ‹master› 
╰─$ cmake . -DARCH=gowin -DGOWIN_BBA_EXECUTABLE=`which gowin_bba`

-- The CXX compiler identification is AppleClang 15.0.0.15000309
-- The C compiler identification is AppleClang 15.0.0.15000309
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Building with IPO
-- Found Python3: /Users/justinzaun/local/brew/Frameworks/Python.framework/Versions/3.11/bin/python3.11 (found suitable version "3.11.8", minimum required is "3.5") found components: Interpreter
-- Found Python3: /Users/justinzaun/local/brew/opt/python@3.11/Frameworks/Python.framework/Versions/3.11/include/python3.11 (found suitable version "3.11.8", minimum required is "3.5") found components: Development Development.Module Development.Embed
-- Found Boost: /Users/justinzaun/local/brew/include (found version "1.84.0") found components: filesystem program_options iostreams system thread regex chrono atomic
-- Found Boost: /Users/justinzaun/local/brew/include (found version "1.84.0") found components: program_options filesystem system
-- Configuring architecture: gowin
-- Enabled Gowin devices: GW1N-1;GW1NZ-1;GW1N-4;GW1N-9;GW1N-9C;GW1NS-2;GW1NS-4;GW2A-18
-- gowin_bba executable: /Users/justinzaun/local/brew/bin/gowin_bba
-- Using Gowin chipdb: /Users/justinzaun/Development/nextpnr/gowin/chipdb
-- Configuring done (1.6s)
-- Generating done (0.0s)
-- Build files have been written to: /Users/justinzaun/Development/nextpnr
```

**Build:**

```
╭─justinzaun@Hogsmeade ~/Development/nextpnr ‹master› 
╰─$ make
[  1%] Generating chipdb/chipdb-GW1N-1.bba
[  2%] Generating chipdb/chipdb-GW1NZ-1.bba
[  3%] Generating chipdb/chipdb-GW1N-4.bba
[  5%] Generating chipdb/chipdb-GW1N-9.bba
[  6%] Generating chipdb/chipdb-GW1N-9C.bba
[  7%] Generating chipdb/chipdb-GW1NS-2.bba
[  8%] Generating chipdb/chipdb-GW1NS-4.bba
[ 10%] Generating chipdb/chipdb-GW2A-18.bba
[ 10%] Built target chipdb-gowin-bbas
[ 11%] Building CXX object bba/CMakeFiles/bbasm.dir/main.cc.o
[ 12%] Linking CXX executable bbasm
[ 12%] Built target bbasm
[ 13%] Generating gowin/chipdb/chipdb-GW1N-1.cc
[ 15%] Generating gowin/chipdb/chipdb-GW1NZ-1.cc
[ 16%] Generating gowin/chipdb/chipdb-GW1N-4.cc
[ 17%] Generating gowin/chipdb/chipdb-GW1N-9.cc
[ 18%] Generating gowin/chipdb/chipdb-GW1N-9C.cc
[ 20%] Generating gowin/chipdb/chipdb-GW1NS-2.cc
[ 21%] Generating gowin/chipdb/chipdb-GW1NS-4.cc
[ 22%] Generating gowin/chipdb/chipdb-GW2A-18.cc
[ 22%] Built target chipdb-gowin-bins
[ 23%] Building CXX object CMakeFiles/chipdb-gowin.dir/gowin/chipdb/chipdb-GW1N-1.cc.o
[ 25%] Building CXX object CMakeFiles/chipdb-gowin.dir/gowin/chipdb/chipdb-GW1NZ-1.cc.o
[ 26%] Building CXX object CMakeFiles/chipdb-gowin.dir/gowin/chipdb/chipdb-GW1N-4.cc.o
[ 27%] Building CXX object CMakeFiles/chipdb-gowin.dir/gowin/chipdb/chipdb-GW1N-9.cc.o
[ 28%] Building CXX object CMakeFiles/chipdb-gowin.dir/gowin/chipdb/chipdb-GW1N-9C.cc.o
[ 30%] Building CXX object CMakeFiles/chipdb-gowin.dir/gowin/chipdb/chipdb-GW1NS-2.cc.o
[ 31%] Building CXX object CMakeFiles/chipdb-gowin.dir/gowin/chipdb/chipdb-GW1NS-4.cc.o
[ 32%] Building CXX object CMakeFiles/chipdb-gowin.dir/gowin/chipdb/chipdb-GW2A-18.cc.o
[ 42%] Built target chipdb-gowin
[ 43%] Building CXX object CMakeFiles/nextpnr-gowin.dir/common/kernel/archcheck.cc.o
[ 45%] Building CXX object CMakeFiles/nextpnr-gowin.dir/common/kernel/basectx.cc.o
[ 46%] Building CXX object CMakeFiles/nextpnr-gowin.dir/common/kernel/bits.cc.o
[ 47%] Building CXX object CMakeFiles/nextpnr-gowin.dir/common/kernel/command.cc.o
[ 48%] Building CXX object CMakeFiles/nextpnr-gowin.dir/common/kernel/context.cc.o
[ 50%] Building CXX object CMakeFiles/nextpnr-gowin.dir/common/kernel/design_utils.cc.o
[ 51%] Building CXX object CMakeFiles/nextpnr-gowin.dir/common/kernel/embed.cc.o
[ 52%] Building CXX object CMakeFiles/nextpnr-gowin.dir/common/kernel/handle_error.cc.o
[ 53%] Building CXX object CMakeFiles/nextpnr-gowin.dir/common/kernel/idstring.cc.o
[ 55%] Building CXX object CMakeFiles/nextpnr-gowin.dir/common/kernel/idstringlist.cc.o
[ 56%] Building CXX object CMakeFiles/nextpnr-gowin.dir/common/kernel/log.cc.o
[ 57%] Building CXX object CMakeFiles/nextpnr-gowin.dir/common/kernel/nextpnr.cc.o
[ 58%] Building CXX object CMakeFiles/nextpnr-gowin.dir/common/kernel/nextpnr_assertions.cc.o
[ 60%] Building CXX object CMakeFiles/nextpnr-gowin.dir/common/kernel/nextpnr_namespaces.cc.o
[ 61%] Building CXX object CMakeFiles/nextpnr-gowin.dir/common/kernel/nextpnr_types.cc.o
[ 62%] Building CXX object CMakeFiles/nextpnr-gowin.dir/common/kernel/property.cc.o
[ 63%] Building CXX object CMakeFiles/nextpnr-gowin.dir/common/kernel/pybindings.cc.o
[ 65%] Building CXX object CMakeFiles/nextpnr-gowin.dir/common/kernel/report.cc.o
[ 66%] Building CXX object CMakeFiles/nextpnr-gowin.dir/common/kernel/sdf.cc.o
[ 67%] Building CXX object CMakeFiles/nextpnr-gowin.dir/common/kernel/str_ring_buffer.cc.o
[ 68%] Building CXX object CMakeFiles/nextpnr-gowin.dir/common/kernel/svg.cc.o
[ 70%] Building CXX object CMakeFiles/nextpnr-gowin.dir/common/kernel/timing.cc.o
[ 71%] Building CXX object CMakeFiles/nextpnr-gowin.dir/common/kernel/timing_log.cc.o
[ 72%] Building CXX object CMakeFiles/nextpnr-gowin.dir/common/place/detail_place_core.cc.o
[ 73%] Building CXX object CMakeFiles/nextpnr-gowin.dir/common/place/parallel_refine.cc.o
[ 75%] Building CXX object CMakeFiles/nextpnr-gowin.dir/common/place/place_common.cc.o
[ 76%] Building CXX object CMakeFiles/nextpnr-gowin.dir/common/place/placer1.cc.o
[ 77%] Building CXX object CMakeFiles/nextpnr-gowin.dir/common/place/placer_heap.cc.o
[ 78%] Building CXX object CMakeFiles/nextpnr-gowin.dir/common/place/placer_static.cc.o
[ 80%] Building CXX object CMakeFiles/nextpnr-gowin.dir/common/place/timing_opt.cc.o
[ 81%] Building CXX object CMakeFiles/nextpnr-gowin.dir/common/route/router1.cc.o
[ 82%] Building CXX object CMakeFiles/nextpnr-gowin.dir/common/route/router2.cc.o
[ 83%] Building CXX object CMakeFiles/nextpnr-gowin.dir/3rdparty/json11/json11.cpp.o
[ 85%] Building CXX object CMakeFiles/nextpnr-gowin.dir/3rdparty/oourafft/fftsg.cc.o
[ 86%] Building CXX object CMakeFiles/nextpnr-gowin.dir/3rdparty/oourafft/fftsg2d.cc.o
[ 87%] Building CXX object CMakeFiles/nextpnr-gowin.dir/json/jsonwrite.cc.o
[ 88%] Building CXX object CMakeFiles/nextpnr-gowin.dir/frontend/json_frontend.cc.o
[ 90%] Building CXX object CMakeFiles/nextpnr-gowin.dir/gowin/arch.cc.o
[ 91%] Building CXX object CMakeFiles/nextpnr-gowin.dir/gowin/arch_pybindings.cc.o
[ 92%] Building CXX object CMakeFiles/nextpnr-gowin.dir/gowin/cells.cc.o
[ 93%] Building CXX object CMakeFiles/nextpnr-gowin.dir/gowin/cst.cc.o
[ 95%] Building CXX object CMakeFiles/nextpnr-gowin.dir/gowin/gfx.cc.o
[ 96%] Building CXX object CMakeFiles/nextpnr-gowin.dir/gowin/globals.cc.o
[ 97%] Building CXX object CMakeFiles/nextpnr-gowin.dir/gowin/main.cc.o
[ 98%] Building CXX object CMakeFiles/nextpnr-gowin.dir/gowin/pack.cc.o
[100%] Linking CXX executable nextpnr-gowin
[100%] Built target nextpnr-gowin
```

**Install:**

```
╭─justinzaun@Hogsmeade ~/Development/nextpnr ‹master●› 
╰─$ DESTDIR=$HOME/local make install                             
[ 10%] Built target chipdb-gowin-bbas
[ 12%] Built target bbasm
[ 22%] Built target chipdb-gowin-bins
[ 42%] Built target chipdb-gowin
[100%] Built target nextpnr-gowin
Install the project...
-- Install configuration: "Release"
-- Installing: /Users/justinzaun/local/usr/local/bin/nextpnr-gowin
```

**Error:**

```
╭─justinzaun@Hogsmeade ~/Development/fpgs/blink 
╰─$     nextpnr-gowin --json synthesis.json --freq 27 --write bitstream.json --device GW2AR-LV18QN88C8/I7 --family GW2AR-18 --cst tangnano20k.cst

dyld[11812]: Symbol not found: __ZNK5boost10filesystem4path7stem_v3Ev
  Referenced from: <0E613CE1-6767-3208-8086-BA66F39495D4> /Users/justinzaun/local/bin/nextpnr-gowin
  Expected in:     <6EB9A896-C802-39D5-B5B8-107A21609B07> /Users/justinzaun/local/brew/Cellar/boost/1.84.0_1/lib/libboost_filesystem-mt.dylib
[1]    11812 abort      nextpnr-gowin --json synthesis.json --freq 27 --write bitstream.json --device
```

I did check and `/Users/justinzaun/local/brew/Cellar/boost/1.84.0_1/lib/libboost_filesystem-mt.dylib` does exist.

yrabbit, 2024-03-30:
This is not a nextpnr-gowin specific bug and probably won't solve your current problem, but please don't use nextpnr-gowin - starting around May 1st I plan to ask for this part to be removed from nextpnr.

Gowin chips are supported via nextpnr-himbaechel:

```
cmake . -DARCH="himbaechel" -DHIMBAECHEL_GOWIN_DEVICES="all"
make -j$(nproc)
sudo make install
```

``` shell
nextpnr-himbaechel --json synthesis.json --write bitstream.json --device GW2AR-LV18QN88C8/I7 --vopt family GW2AR-18 --vopt cst tangnano20k.cst
```

Ravenslofty, 2024-03-30:
Also, this seems like the same problem as #1297 and #1292: there's a problem with your boost libraries.

zaun, 2024-03-30:
@yrabbit - no worries, I'm happy to move over to  nextpnr-himbaechel. I've got it compiled and am trying to run the `nextpnr-himbaechel` but am getting an error about the chipdb not being found:

```
nextpnr-himbaechel --json synthesis.json --write bitstream.json --device GW2AR-LV18QN88C8/I7 --vopt family GW2AR-18 --vopt cst tangnano20k.cst
Info: Using uarch 'gowin' for device 'GW2AR-LV18QN88C8/I7'
ERROR: Unable to read chipdb /Users/justinzaun/local/bin/../share/nextpnr//himbaechel/gowin/chipdb-.bin
0 warnings, 1 error
make: *** [bitstream.json] Error 255
╭─justinzaun@Hogsmeade ~/Development/fpgs/blink 
╰─$ ls /Users/justinzaun/local/bin/../share/nextpnr//himbaechel/gowin/                      2 ↵
chipdb-GW1N-1.bin   chipdb-GW1N-9.bin   chipdb-GW1NS-4.bin  chipdb-GW2A-18.bin
chipdb-GW1N-4.bin   chipdb-GW1N-9C.bin  chipdb-GW1NZ-1.bin  chipdb-GW2A-18C.bin
```

The build and install went fine:

```
╭─justinzaun@Hogsmeade ~/Development/nextpnr ‹master●› 
╰─$ cmake . -DARCH="himbaechel" -DHIMBAECHEL_GOWIN_DEVICES="all" -DCMAKE_INSTALL_PREFIX=/ 
make -j$(nproc)

-- Building with IPO
-- Found Python3: /Users/justinzaun/local/brew/opt/python@3.11/Frameworks/Python.framework/Versions/3.11/include/python3.11 (found suitable version "3.11.8", minimum required is "3.5") found components: Development Development.Module Development.Embed
-- Found Boost: /Users/justinzaun/local/brew/include (found version "1.84.0") found components: filesystem program_options iostreams system thread regex chrono atomic
-- Found Boost: /Users/justinzaun/local/brew/include (found version "1.84.0") found components: program_options filesystem system
-- Configuring architecture: himbaechel
-- Configuring Himbaechel-Example uarch
-- Configuring Himbaechel-Gowin uarch
-- Found Python3: /Users/justinzaun/local/brew/Frameworks/Python.framework/Versions/3.11/bin/python3.11 (found suitable version "3.11.8", minimum required is "3.5") found components: Interpreter
-- Enabled Himbaechel-Gowin devices: GW1N-1;GW1NZ-1;GW1N-4;GW1N-9;GW1N-9C;GW1NS-4;GW2A-18;GW2A-18C
-- Apycula install prefix: (using system Python)
-- Configuring Xilinx uarch
-- Enabled Himbaechel-Xilinx devices: 
-- Configuring done (0.7s)
-- Generating done (0.0s)
-- Build files have been written to: /Users/justinzaun/Development/nextpnr
zsh: command not found: nproc
[  0%] Built target chipdb-himbaechel-xilinx
[  2%] Built target bbasm
[ 14%] Built target chipdb-himbaechel-gowin
[100%] Built target nextpnr-himbaechel
╭─justinzaun@Hogsmeade ~/Development/nextpnr ‹master●› 
╰─$ DESTDIR=$HOME/local make install                                                     
[ 85%] Built target nextpnr-himbaechel
[ 88%] Built target bbasm
[100%] Built target chipdb-himbaechel-gowin
[100%] Built target chipdb-himbaechel-xilinx
Install the project...
-- Install configuration: "Release"
-- Installing: /Users/justinzaun/local/bin/nextpnr-himbaechel
-- Installing: /Users/justinzaun/local/share/nextpnr/himbaechel/gowin
-- Installing: /Users/justinzaun/local/share/nextpnr/himbaechel/gowin/chipdb-GW1N-9.bin
-- Installing: /Users/justinzaun/local/share/nextpnr/himbaechel/gowin/chipdb-GW1N-9C.bin
-- Installing: /Users/justinzaun/local/share/nextpnr/himbaechel/gowin/chipdb-GW1NS-4.bin
-- Installing: /Users/justinzaun/local/share/nextpnr/himbaechel/gowin/chipdb-GW2A-18.bin
-- Installing: /Users/justinzaun/local/share/nextpnr/himbaechel/gowin/chipdb-GW2A-18C.bin
-- Installing: /Users/justinzaun/local/share/nextpnr/himbaechel/gowin/chipdb-GW1N-1.bin
-- Installing: /Users/justinzaun/local/share/nextpnr/himbaechel/gowin/chipdb-GW1NZ-1.bin
-- Installing: /Users/justinzaun/local/share/nextpnr/himbaechel/gowin/chipdb-GW1N-4.bin
-- Installing: /Users/justinzaun/local/share/nextpnr/himbaechel/xilinx
```

I tried `GW2A-18` rather than `GW2AR-18` but the same thing happens:

```
nextpnr-himbaechel --json synthesis.json --write bitstream.json --device GW2A-LV18QN88C8/I7 --vopt family GW2A-18 --vopt cst tangnano20k.cst 
Info: Using uarch 'gowin' for device 'GW2A-LV18QN88C8/I7'
ERROR: Unable to read chipdb /Users/justinzaun/local/bin/../share/nextpnr//himbaechel/gowin/chipdb-.bin
0 warnings, 1 error
```

yrabbit, 2024-03-30:
I wonder where I got it wrong.
Try 
```
--device GW2AR-LV18QN88C8/I7 --vopt family=GW2A-18C
```

If it doesn’t work, then I’ll compile the latest nextpnr and see what’s wrong.

zaun, 2024-03-30:
@yrabbit - Everything is working.  Thank you for the help!

closed 2024-03-30 by zaun