YosysHQ/nextpnr #1307: DSP Primitives not working on GOWIN or Himbaechel

bjoernskau, 2024-04-03:
### Hello, and thank you on advance.

We are making a FIR filter on the tangnano9k, and wants to use the Macro for the Module "MULT36x36".
We are using Yosys to synthesize our code, which works perfectly, when we try to place and route using nextpnr, we get the following error:

`Info: Placed 1 cells based on constraints.
ERROR: Unable to place cell 'mu_0', no BELs remaining to implement cell type 'MULT36X36'
0 warnings, 1 error`

We are not sure that the DSP Primitives for the Gowin is supported yet, but is seems to be a newly added feature??

We have tried both using Gowin:
`nextpnr-gowin --json $(PRO).json --freq 27 --write $(PRO)_pnr.json --device ${DEVICE} --family ${FAMILY} --cst ${BOARD}.cst`

And Himbaechel:
`nextpnr-himbaechel --json $(PRO).json --write $(PRO)_pnr.json \
	--device ${DEVICE} --vopt family=${FAMILY} --vopt cst=$(BOARD_DIR).cst`

This works normally, and we primarily use Himbaechel because we use the BSRAM macros aswell.

**Is the DSP primitives fully intergrated in Himbaechel, if so how do we use it?**

_We would also like to thank you all for your good work at YosysHQ, we have been using your opensource tools for about 6 months now, and are really looking forward to what that team might bring in the future, keep up the good work._

Best regards,
Kasper and his team.

yrabbit, 2024-04-03:
The latest versions of apicula and nextpnr-himbaechel have support for MULT36X36.

You need a combination of apicula v 0.12 and nextpnr master branch.

bjoernskau, 2024-04-03:
I made a "git pull" on all repos, and reinstalled apicula and nextpnr-himbaechel using the provided readme, did these steps:

`pip install apycula`

`which gowin_bba # check if binaries are on the path
python -m site --user-base # find the site packages base directory
ls $HOME/.local/bin # confirm the binaries are installed in this folder
export PATH="$HOME/.local/bin:$PATH" # add binaries to the path`

and for himbaechel:
`cmake . -DARCH="himbaechel" -DHIMBAECHEL_GOWIN_DEVICES="all"
make -j$(nproc)
sudo make install`

is there anything else I need to do?

yrabbit, 2024-04-03:
Well, you can download the apicula sources to get a directory with examples and try to compile the example with MULT36X36. If it doesn't work, then I would like to see the logs

``` shell
git clone git@github.com:YosysHQ/apicula.git
cd apicula/examples/himbaechel
gmake -f Makefile.himbaechel dsp-mult36x36-tangnano9k.fs
```

bjoernskau, 2024-04-03:
i corrected the first line to: `git clone https://github.com/YosysHQ/apicula.git`

I attached the log from the terminal. Pls take a look, a thank you for you'r patience. 

[log.txt](https://github.com/YosysHQ/nextpnr/files/14851512/log.txt)

yrabbit, 2024-04-03:
nextpnr is compiled using old apicula databases.
Hmm... I wonder how this happens...
Can you see how many chip database copies you have for Tangnano9k?

```shell
find / -name GW1N-9C.pickle
```

yrabbit, 2024-04-03:
one.
Let's see if it's the same...

```shell
python
import importlib.resources
import pickle
with gzip.open(importlib.resources.files("apycula").joinpath("GW1N-9C.pickle"), 'rb') as f:
      db = pickle.load(f)
print(db.grid[18][1].bels.keys())
```
new base has MULT36X36
[image]

bjoernskau, 2024-04-03:
I first had to install **python-is-python3**, in order to run **"python"** without the **"3"**, maybe that was a problem, because many of your files uses python?? and are run by **"python"**?

I also had to import **gzip** in order for it to work, look at the picture i send.

[image]

yrabbit, 2024-04-03:
All these are correct remarks, but for now we see that you have an old apicula.
It’s already late at night and I can’t figure out how you get the old version. 
We'll continue tomorrow unless someone from another hemisphere tells us what's going on first :)

bjoernskau, 2024-04-03:
Haha, that sounds like a plan, I will keep digging a bit.
I am in Europe so wont be on until 9 CET.

Thanks, have a good night!

yrabbit, 2024-04-03:
This is a bold move!
Yes, now you have a new image packer for fpga, but you don’t have the databases themselves because the process of generating them is somewhat demanding.
The simplest thing you can do is download the ready-made chip base for Tangnano9k and put it in the place where it should be:
https://github.com/YosysHQ/apicula/actions/runs/8498170454/artifacts/1372706469

Ready databases for all chips are located at:
https://github.com/YosysHQ/apicula/actions/runs/8498170454#artifacts

bjoernskau, 2024-04-04:
```
Info: Device utilisation:
Info:                    VCC:     1/    1   100%
Info:                    IOB:     8/  274     2%
Info:                   LUT4:     0/ 8640     0%
Info:                 OSER16:     0/   80     0%
Info:                 IDES16:     0/   80     0%
Info:               IOLOGICI:     0/  276     0%
Info:               IOLOGICO:     0/  276     0%
Info:              MUX2_LUT5:     0/ 4320     0%
Info:              MUX2_LUT6:     0/ 2160     0%
Info:              MUX2_LUT7:     0/ 1080     0%
Info:              MUX2_LUT8:     0/ 1080     0%
Info:                    ALU:     0/ 6480     0%
Info:                    GND:     1/    1   100%
Info:                    DFF:     0/ 6480     0%
Info:              RAM16SDP4:     0/  270     0%
Info:                  BSRAM:     0/   26     0%
Info:                 ALU54D:     0/   10     0%
Info:        MULTADDALU18X18:     0/   10     0%
Info:           MULTALU18X18:     0/   10     0%
Info:           MULTALU36X18:     0/   10     0%
Info:              MULT36X36:     1/    5    20%
Info:              MULT18X18:     0/   20     0%
Info:                MULT9X9:     0/   40     0%
Info:                 PADD18:     0/   20     0%
Info:                  PADD9:     0/   40     0%
Info:                    GSR:     1/    1   100%
Info:                    OSC:     0/    1     0%
Info:                   rPLL:     0/    2     0%
Info:                   BUFG:     0/   22     0%
```

I think you know what this means!
Thank you, a lot!!

I was forced to add all .pickle filles to the folder, fx:
`FileNotFoundError: [Errno 2] No such file or directory: '/home/kasper/.local/lib/python3.10/site-packages/apycula/GW2A-18C.pickle'
`

So when I downloaded all of them and added them to the path above, then I could run the:
`make`
After the cmake:
`cmake . -DARCH="himbaechel" -DHIMBAECHEL_GOWIN_DEVICES="all"
`
And the after the `sudo make install` I tried to run my project with the following commands:

`yosys -p "read_verilog mult36.v top.v; synth_gowin -json top.json"`
( I made the mult36 in my own Module)
`nextpnr-himbaechel --json top.json --write pnrtop.json --device GW1NR-LV9QN88PC6/I5 --vopt family=GW1N-9C --vopt cst=tangnano9k.cst`

And then it finished normally with 0 errors!

Thank you, a lot, for the help and fast response on my issue. Maybe this thread can help someone dealing with the same issue.

Keep up the good work, we are looking forward to future updates on this opensource platform, we love it!

I can now continue testing my DSP FIR filter in real-time using the MULT36x36 and the ALU54D as an accumulator.

PS: Do you guys have a Discord/Forum where I/my team can join discussions and ask simple questions?

See you!

yrabbit, 2024-04-04:
Great!
You can ask a question about apicula in the chat as indicated
https://github.com/YosysHQ/apicula#resources