YosysHQ/nextpnr #1317: Routing fails on Crosslink Nexus: "Failed to route arc 0.0 of net 'clk'"

polymerizedsage, 2024-04-29:
Struggling with an issue where nextpnr fails while trying to build a design for the CrosslinkNX-40. The full error message is: 
`ERROR: Failed to route arc 0.0 of net 'clk', from X86/Y53/JCLKOP_PLL_CORE_I_PLL_LMMI to X49/Y56/JCLKI_DCC_DCC2.`

I am trying to narrow down my design to a minimum reproducible example but its proving tricky. The design I am working on is for an I2S controller and the issue seems to have something to do with how I am dividing the system clock to the serial bus clock. I am using the Crosslink PLL to generate a 64MHz system clock which I divide to a 4MHz serial clock. 

The TX and RX modules build just fine on their own, but when I try and use both of them in the same design I get the error above. Ill keep working toward something concise that reproduces the issue, but I am finding it hard to grasp which piece of my design is the one actually causing the error. 

I should add, the design is synthesized and place and routed with the Radiant toolchain just fine. So I am pretty confident this is not a fundamental issue with my HDL but some interaction between the HDL and the tool.

gatecat, 2024-05-03:
Usually these bugs don't require a minimal reproducer to trace down - so if you are able to share your full design, that will also be useful.

polymerizedsage, 2024-05-16:
The top level design that fails to build can be found here: https://github.com/girlvoice/girlvoice-gateware/blob/main/girlvoice/lifcl_evn_top.py 

The basic instructions in the readme should hopefully be enough to reproduce. For the record. I just re-encountered this bug trying to build a Litex core for a different board that also uses a CrosslinkNX. It would be nice to get this figured out.

gatecat, 2024-05-16:
Thanks, can you confirm that https://github.com/YosysHQ/nextpnr/pull/1325 works for you too?

polymerizedsage, 2024-05-17:
Yep!

PR #1325 (2024-05-17): nexus: Use a toposort when preplacing clock primitives
Fixes #1317
closed 2024-05-17 by gatecat