YosysHQ/nextpnr #1320: ecp5: trouble using IO_TYPE=LVDS

Ownasaurus, 2024-05-06:
I am trying to use an LVDS differential pair as input. In my .LPF file, I have the following:

```
LOCATE COMP "tmds_c_p" SITE "F2";
IOBUF PORT "tmds_c_p" PULLMODE=UP IO_TYPE=LVDS;
LOCATE COMP "tmds_c_n" SITE "E2";
IOBUF PORT "tmds_c_n" PULLMODE=UP IO_TYPE=LVDS;
```

However, I unfortunately get the following message:
`ERROR: cannot place differential IO at location PIOD`

Any idea why this might be happening? I believe F2 and E2 should be appropriate for a differential signal as input. Bank 7, corresponding C/D pins. Am I incorrectly setting the constraints in the LPF file?

(Note the signals are named TMDS but are being converted to the appropriate LVDS logic levels before being passed to the FPGA.)

Thank you very much for your time.

whitequark, 2024-05-06:
Try using just the positive pin, toolchains *usually* only care about that half.

Ownasaurus, 2024-05-06:
> Try using just the positive pin, toolchains _usually_ only care about that half.

Thanks. Then, my top module would just have one input which would be the proper signal? I wouldn't have to explicitly put the T/+ and C/- signals through an ILVDS primitive?

Thanks!

whitequark, 2024-05-06:
> Then, my top module would just have one input which would be the proper signal?

That's correct.

Ownasaurus, 2024-05-06:
Thank you very much! It certainly synthesizes this way -- just constraining the T/+ part of the pair and only having one input to my top module. I will not be able to test for data integrity for a few weeks until my custom hardware has arrived. I will re-open or create a new issue if I have any further questions.

Thanks again!

closed 2024-05-06 by Ownasaurus

whitequark, 2024-05-06:
You can check it in the nextpnr GUI. We also do this in Amaranth for a while and it seems to work fine in every case it's been tested.

Ownasaurus, 2024-05-07:
> You can check it in the nextpnr GUI. We also do this in Amaranth for a while and it seems to work fine in every case it's been tested.

Thank you for your advice, whitequark. I had never tried the GUI before! I rebuilt nextpnr with the GUI, and tried to figure out how to use it. Is there a good source of documentation for the GUI?

Anyhow, I I think I figured it out well enough to do some basic analysis. In this test, it looks like it assigned tmds_c to BEL X0/Y44/PIOC. I do not see anything assigned to the corresponding PIOD. Would I expect to see the complement assigned there automatically?

whitequark, 2024-05-07:
> Would I expect to see the complement assigned there automatically?

That's actually a good question, I would assume yes but I've never personally done that before on ECP5.

Ownasaurus, 2024-05-07:
> > Would I expect to see the complement assigned there automatically?
> 
> That's actually a good question, I would assume yes but I've never personally done that before on ECP5.

Well, I built the soc_versa5g example from the prjtrellis repo. It uses an LVDS clkin. the nextpnr gui shows the same behavior as mine: the T/+ is routed but the C/- is **not**. So I am going to assume this is a success and working as intended. Thanks again for your assistance. And good luck with your work on amaranth and such! I hope somebody else finds this discussion helpful.