YosysHQ/nextpnr #1321: Does anyone know how to get this working in Void Linux?

Xeraster, 2024-05-10:
I get this error when running make:
`
CMake Error at /usr/share/cmake-3.27/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Python3 (missing: Python3_INCLUDE_DIRS Development
  Development.Module Development.Embed) (found suitable version "3.12",
  minimum required is "3.5")
Call Stack (most recent call first):
  /usr/share/cmake-3.27/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.27/Modules/FindPython/Support.cmake:3824 (find_package_handle_standard_args)
  /usr/share/cmake-3.27/Modules/FindPython3.cmake:545 (include)
  CMakeLists.txt:199 (find_package)

`

I spent all afternoon trying to either compile nextpnr or compile python 3.5 from source but was ultimately unsuccessful. I even tried modifying the scripts that throw an error when python 3.5 isn't found. This didn't work and I don't know why. I think next i'm going to try static complication: that's where you compile the program in such a way that all the dependencies get bundled into the binary. This makes the binary larger in size but maybe it will work.

whitequark, 2024-05-11:
If you have working Python, you could try installing [the YoWASP distribution](https://yowasp.org):

```
pip install yowasp-nextpnr-ice40
yowasp-nextpnr-ice40 --help
```

gatecat, 2024-05-11:
The failure here is that it can't find Python with a minimum version of 3.5 (actually, this is probably out of date and newer is required now) and the right development headers - 3.12, as it's finding,  is definitely fine too just the development parts are missing - probably the relevant development package from your distribution must be installed.

closed 2024-09-26 by gatecat