YosysHQ/nextpnr #1323: Matching Bel type shows as not matching

KelvinChung2000, 2024-05-14:
I am experimenting with my architecture with `Himachal`. I am now trying to route a toy example and get things working first before increasing the complexity of the design. 

I synthesised the following module with my top design:
```
(* blackbox *)
module IOB (input T, I, output O);
endmodule

(* blackbox *)
module Global_Clock (output CLK);
endmodule

module top_wrapper;

  wire [27:0] io_in, io_out, io_oeb;
  (* BEL="X1Y0/IO27" *) IOB io27_i (.O(io_in[27]), .I(io_out[27]), .T(io_oeb[27]));
  (* BEL="X1Y0/IO26" *) IOB io26_i (.O(io_in[26]), .I(io_out[26]), .T(io_oeb[26]));
  (* BEL="X1Y0/IO25" *) IOB io25_i (.O(io_in[25]), .I(io_out[25]), .T(io_oeb[25]));
  (* BEL="X1Y0/IO24" *) IOB io24_i (.O(io_in[24]), .I(io_out[24]), .T(io_oeb[24]));
  (* BEL="X1Y0/IO23" *) IOB io23_i (.O(io_in[23]), .I(io_out[23]), .T(io_oeb[23]));
  (* BEL="X1Y0/IO22" *) IOB io22_i (.O(io_in[22]), .I(io_out[22]), .T(io_oeb[22]));
  (* BEL="X1Y0/IO21" *) IOB io21_i (.O(io_in[21]), .I(io_out[21]), .T(io_oeb[21]));
  (* BEL="X1Y0/IO20" *) IOB io20_i (.O(io_in[20]), .I(io_out[20]), .T(io_oeb[20]));
  (* BEL="X1Y0/IO19" *) IOB io19_i (.O(io_in[19]), .I(io_out[19]), .T(io_oeb[19]));
  (* BEL="X1Y0/IO18" *) IOB io18_i (.O(io_in[18]), .I(io_out[18]), .T(io_oeb[18]));
  (* BEL="X1Y0/IO17" *) IOB io17_i (.O(io_in[17]), .I(io_out[17]), .T(io_oeb[17]));
  (* BEL="X1Y0/IO16" *) IOB io16_i (.O(io_in[16]), .I(io_out[16]), .T(io_oeb[16]));
  (* BEL="X1Y0/IO15" *) IOB io15_i (.O(io_in[15]), .I(io_out[15]), .T(io_oeb[15]));
  (* BEL="X1Y0/IO14" *) IOB io14_i (.O(io_in[14]), .I(io_out[14]), .T(io_oeb[14]));
  (* BEL="X1Y0/IO13" *) IOB io13_i (.O(io_in[13]), .I(io_out[13]), .T(io_oeb[13]));
  (* BEL="X1Y0/IO12" *) IOB io12_i (.O(io_in[12]), .I(io_out[12]), .T(io_oeb[12]));
  (* BEL="X1Y0/IO11" *) IOB io11_i (.O(io_in[11]), .I(io_out[11]), .T(io_oeb[11]));
  (* BEL="X1Y0/IO10" *) IOB io10_i (.O(io_in[10]), .I(io_out[10]), .T(io_oeb[10]));
  (* BEL="X1Y0/IO9" *) IOB io9_i (.O(io_in[9]), .I(io_out[9]), .T(io_oeb[9]));
  (* BEL="X1Y0/IO8" *) IOB io8_i (.O(io_in[8]), .I(io_out[8]), .T(io_oeb[8]));
  (* BEL="X1Y0/IO7" *) IOB io7_i (.O(io_in[7]), .I(io_out[7]), .T(io_oeb[7]));
  (* BEL="X1Y0/IO6" *) IOB io6_i (.O(io_in[6]), .I(io_out[6]), .T(io_oeb[6]));
  (* BEL="X1Y0/IO5" *) IOB io5_i (.O(io_in[5]), .I(io_out[5]), .T(io_oeb[5]));
  (* BEL="X1Y0/IO4" *) IOB io4_i (.O(io_in[4]), .I(io_out[4]), .T(io_oeb[4]));
  (* BEL="X1Y0/IO3" *) IOB io3_i (.O(io_in[3]), .I(io_out[3]), .T(io_oeb[3]));
  (* BEL="X1Y0/IO2" *) IOB io2_i (.O(io_in[2]), .I(io_out[2]), .T(io_oeb[2]));
  (* BEL="X1Y0/IO1" *) IOB io1_i (.O(io_in[1]), .I(io_out[1]), .T(io_oeb[1]));
  (* BEL="X1Y0/IO0" *) IOB io0_i (.O(io_in[0]), .I(io_out[0]), .T(io_oeb[0]));

  wire clk;
  (* keep *) Global_Clock clk_i (.CLK(clk));
  assign io_oeb = 28'b1;

  top top_i(.clk(clk),
            .a(io_in[7:0]),
            .b(io_in[15:8]),
            .dst(io_out[23:16]));

endmodule
```

When I run `nextpnr-himbaechel`, I get the following error:
```
ERROR: Bel 'X1Y0/IO4' of type 'IOB' does not match cell 'io4_i' of type 'IOB'
```
Am I synthesising my design wrong? Or am I missing something?

yrabbit, 2024-05-14:
Did you happen to override the isValidBelForCellType() function in an interesting way?

KelvinChung2000, 2024-05-14:
I just checked the `example.cc`  in the source is overriding it. That might be why.