YosysHQ/nextpnr #1330: Could not use nextpnr-gowin with --family GW2A-18C

janeoa, 2024-06-20:
I am using m1 mac with macos and trying to run 
```bash
nextpnr-gowin --json uart.json --write uart_pnr.json --freq 27 --device GW2A-LV18PG256C8/I7 --family GW2A-18C --cst tangprimer20k.cst 
```

I have build the nextpnr from this git using
```bash
cmake -DARCH=gowin -DBUILD_GUI=OFF -DARCH="himbaechel" -DHIMBAECHEL_GOWIN_DEVICES="all" -DPYTHON_EXECUTABLE=~/dev/oss-cad-suite/py3bin/python3 .

make -j$(nproc)

sudo make install
```

and now I have following files
```bash
ls /usr/local/share/nextpnr/himbaechel/gowin/
chipdb-GW1N-1.bin	chipdb-GW1N-4.bin	chipdb-GW1N-9.bin	chipdb-GW1N-9C.bin	chipdb-GW1NS-4.bin	chipdb-GW1NZ-1.bin	chipdb-GW2A-18.bin	chipdb-GW2A-18C.bin
```

But when I envoke nextpnr-gowin for GW2A-18C, it fails
```
nextpnr-gowin --json uart.json --write uart_pnr.json --freq 27 --device GW2A-LV18PG256C8/I7 --family GW2A-18C --cst tangprimer20k.cst
ERROR: Failed to load chipdb 'gowin/chipdb-GW2A-18C.bin'
0 warnings, 1 error
```

what am I missing?

yrabbit, 2024-06-20:
- You still run nextpnt-gowin, but now you need to use Himbaechel. 
  - Tangprimer20k is not GW2A-18C, it's just GW2A-18

``` shell
#Tangnano20k
nextpnr-himbaechel  --json  uart.json --write  uart_pnr.json --device GW2AR-LV18QN88C8/I7 --vopt family=GW2A-18C --vopt cst=tangnano20k. cst

#Tangprimer20k
nextpnr-himbaechel  --json  uart.json --write  uart_pnr.json --device GW2A-LV18PG256C8/I7 --vopt family=GW2A-18 --vopt cst=primer20k.cst
```

yrabbit, 2024-06-24:
If everything worked out for you, then close this issue, please. :smile:

closed 2024-07-11 by janeoa