YosysHQ/nextpnr #1331: Is there MachXO3 support?

philtomson, 2024-06-21:
I see that there's a nextpnr-machxo2 but no nextpnr-machxo3. I thought that maybe I could try using nextpnr-machxo2 with a MachXO3 target device like:

$ nextpnr-machxo2 --device LCMXO3L-2100E-5MG121C serial_multiplier.blif --textcfg serial_multiplier_cfg.txt

But that results in:
ERROR: Unsupported MachXO2 chip type.

(Version nextpnr-0.7-28-gedcafcf0)

gatecat, 2024-06-22:
@mmicko ?

mmicko, 2024-06-22:
@philtomson by default only XO2 1200 series and XO3 6900 series are active ( MACHXO2_DEVICES parameter ). You can enable this specific device by adding -DMACHXO2_DEVICES="2100" to cmake parameters.

Idea is to make possible to enable all of these, but issue is that databases are big and takes a while to generate.

@gatecat maybe we should start porting this to himbaechel ?