YosysHQ/nextpnr #1335: An option to reject inputs where some of the clock nets have no defined or derived constraints

whitequark, 2024-06-29:
Due to a combination of factors, I've realized recently that I've been building Glasgow bitstreams with the default 12 MHz constraint, which of course did not work at 80 MHz post-PLL.

We have `--pcf-allow-unconstrained`, I propose `--clk-allow-unconstrained` and similarly switching the default (or at the very least, `--clk-deny-unconstrained` to be available as an option).

rowanG077, 2024-06-29:
I would rather propose making it an unconstrained clock a warning. And then you can enable warning as error.

whitequark, 2024-06-30:
> I would rather propose making an unconstrained clock a warning.

When would you ever want an unconstrained clock in a real design?

Unconstrained pins are reasonably common, like if you haven't done PCB layout yet. Unconstrained clocks essentially result in garbage-in, garbage-out, unless your clock is 12 MHz or less, which is fairly uncommon in amaranth-boards.

Also, I'm not sure if the default 12 MHz frequency of unconstrained clocks propagates through PLLs, which can be even worse.

rowanG077, 2024-06-30:
I agree that you never want an unconstrained clock in a real design. But breaking peoples builds for relying on something that is bad practice but works is kinda heavy handed. A warning on the short term, maybe with a notice the warning will be upgraded to an error in the future will at least give people time to move.

whitequark, 2024-06-30:
> A warning on the short term, maybe with a notice the warning will be upgraded to an error in the future will at least give people time to move.

That's fine by me. What I want to avoid is any situation where a substring is matched in the warning to elevate it to an error, like Yosys does. It should be a dedicated option.