YosysHQ/nextpnr #1339: Fmax estimation implemented with nextpnr-himbaechel for Gowin devices?

janschiefer, 2024-07-06:
nextpnr-gowin seems to have been removed from oss-cad-suite (current git version), so I'm using nextpnr-himbaechel.

Is Fmax estimation (yet) implemented in nextpnr-himbaechel for Gowin devices?
If yes, how do I enable it?

nextpnr-himbaechel --detailed-timing-report --json build.yosys.json --write build.nextpnr.json --device GW1NR-LV9QN88PC6/I5 --vopt family=GW1N-9C --vopt cst=tangnano9k.cst
Output:
```
(...)
Info: No Fmax available; no interior timing paths found in design.
Info: Checksum: 0xab4fa0f8
Info: Routing globals...
Info:     routed net 'toplevel.clk' using global resources only.

Info: Routing..
(...)

Info: No Fmax available; no interior timing paths found in design.

Info: Program finished normally.
```

Thank you!

yrabbit, 2024-07-06:
I don't think this functionality can be enabled at the moment.
My theoretical knowledge on this subject tends to zero, so we are waiting for volunteers to implement Fmax in himbaechel-gowin :)

janschiefer, 2024-07-07:
So I guess I'll compile nextpnr-gowin manually additionally to using oss-cad-suite for now.

However, thank you!

Ravenslofty, 2024-07-08:
So, there's the lofty/himbaechel-gowin-timing branch (which admittedly I need to revise), but the challenge is that the Fmax numbers produced by it seem way too high compared to the vendor toolchain numbers.

I probably should attempt to debug that and figure out the difference...

closed 2024-09-26 by gatecat