YosysHQ/nextpnr #1343: Apycula not found

ReubenkThomas, 2024-07-27:
I have tried to run the make command after running `cmake . -DARCH="himbaechel" -DHIMBAECHEL_GOWIN_DEVICES="all" -DBUILD_GUI=ON`, after ensuring all the libraries are correctly installed. `apycula` was also installed through `pip install apycula`, which installed it in the (non Homebrew) Python3.12 version. However, I get the following error

```
reubenkthomas@Reubens-MacBook-Air nextpnr % make -j$(nproc)
[  0%] Built target QtPropertyBrowser_autogen_timestamp_deps
[  0%] Built target chipdb-himbaechel-xilinx
[  0%] Built target gui_himbaechel_autogen_timestamp_deps
[  0%] Built target chipdb-himbaechel-example
[  2%] Built target bbasm
[  2%] Built target QtPropertyBrowser_autogen
[  3%] Built target gui_himbaechel_autogen
[  6%] Generating ../../../share/himbaechel/gowin/chipdb-GW1N-4.bin
[  6%] Generating ../../../share/himbaechel/gowin/chipdb-GW1N-1.bin
[  6%] Generating ../../../share/himbaechel/gowin/chipdb-GW1N-9.bin
[  6%] Generating ../../../share/himbaechel/gowin/chipdb-GW1NS-4.bin
[  7%] Generating ../../../share/himbaechel/gowin/chipdb-GW1NZ-1.bin
[  8%] Generating ../../../share/himbaechel/gowin/chipdb-GW1N-9C.bin
[ 19%] Built target QtPropertyBrowser
[ 20%] Generating ../../../share/himbaechel/gowin/chipdb-GW2A-18.bin
Traceback (most recent call last):
  File "/Users/reubenkthomas/Documents/nextpnr/himbaechel/uarch/gowin/gowin_arch_gen.py", line 12, in <module>
    from apycula import chipdb
ModuleNotFoundError: No module named 'apycula'
Traceback (most recent call last):
  File "/Users/reubenkthomas/Documents/nextpnr/himbaechel/uarch/gowin/gowin_arch_gen.py", line 12, in <module>
    from apycula import chipdb
ModuleNotFoundError: No module named 'apycula'
Traceback (most recent call last):
  File "/Users/reubenkthomas/Documents/nextpnr/himbaechel/uarch/gowin/gowin_arch_gen.py", line 12, in <module>
    from apycula import chipdb
ModuleNotFoundError: No module named 'apycula'
Traceback (most recent call last):
  File "/Users/reubenkthomas/Documents/nextpnr/himbaechel/uarch/gowin/gowin_arch_gen.py", line 12, in <module>
    from apycula import chipdb
Traceback (most recent call last):
  File "/Users/reubenkthomas/Documents/nextpnr/himbaechel/uarch/gowin/gowin_arch_gen.py", line 12, in <module>
    from apycula import chipdb
ModuleNotFoundError: No module named 'apycula'
ModuleNotFoundError: No module named 'apycula'
Traceback (most recent call last):
  File "/Users/reubenkthomas/Documents/nextpnr/himbaechel/uarch/gowin/gowin_arch_gen.py", line 12, in <module>
    from apycula import chipdb
ModuleNotFoundError: No module named 'apycula'
make[2]: *** [share/himbaechel/gowin/chipdb-GW1N-1.bin] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [share/himbaechel/gowin/chipdb-GW1N-4.bin] Error 1
make[2]: *** [share/himbaechel/gowin/chipdb-GW1N-9C.bin] Error 1
make[2]: *** [share/himbaechel/gowin/chipdb-GW1N-9.bin] Error 1
make[2]: *** [share/himbaechel/gowin/chipdb-GW1NS-4.bin] Error 1
make[2]: *** [share/himbaechel/gowin/chipdb-GW1NZ-1.bin] Error 1
Traceback (most recent call last):
  File "/Users/reubenkthomas/Documents/nextpnr/himbaechel/uarch/gowin/gowin_arch_gen.py", line 12, in <module>
    from apycula import chipdb
ModuleNotFoundError: No module named 'apycula'
make[2]: *** [share/himbaechel/gowin/chipdb-GW2A-18.bin] Error 1
make[1]: *** [himbaechel/uarch/gowin/CMakeFiles/chipdb-himbaechel-gowin.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 47%] Built target gui_himbaechel
make: *** [all] Error 2
```

yrabbit, 2024-07-27:
Before you did make, you ran cmake, which wrote a lot of stuff on the screen, including what Python it found and would use. It would be interesting to look at this log. Unfortunately, not for me - I'm not an expert on either cmake or Mac.

ReubenkThomas, 2024-07-29:
Hi, just to report back. This issue is mainly caused by having two different Python versions, one Python 3.12 that was downloaded through an installer and has all the pip libraries targeted to, and. the one installed through homebrew when installing the dependencies for nextpnr. To get rid of the issue, you can run:

`brew uninstall --ignore-dependencies python`

which should uninstall the homebrew python version, keep your other python versions, while not removing other libraries you need.

closed 2024-07-29 by ReubenkThomas