YosysHQ/nextpnr #1355: Build is broken

pointcheck, 2024-08-22:
Building from latest source tree gets to many errors like:
```
/tmp/ccg7F5vc.ltrans19.ltrans.o: In function `nextpnr_generic::CommandHandler::executeMain(std::unique_ptr<nextpnr_generic::Context, std::default_delete<nextpnr_generic::Context> >) [clone .constprop.1366]':
/home/rz/RISCV/nextpnr/common/kernel/command.cc:639: undefined reference to `nextpnr_generic::Context::read_sdc(std::istream&)'
collect2: error: ld returned 1 exit status
CMakeFiles/nextpnr-generic.dir/build.make:890: recipe for target 'nextpnr-generic' failed
```

It is not only for "generic" but also for "gowin" and other FPGA targets.

mmicko, 2024-08-22:
Since new files are added and we do not explicitly add files to build scripts, just run `cmake .` and then `make` again and it should build just fine

pointcheck, 2024-08-22:
Indeed, doing cmake fixed this issue. Thanks.

closed 2024-08-22 by pointcheck