YosysHQ/nextpnr #1356: Sequential logic is stall when generated with nextpnr-gowin

pointcheck, 2024-08-26:
It seems ```nextpnr-gowin``` is being broken for quite a while, in bistream generated with ```nextpnr-gowin``` all sequential logic is stall, only combinational logic works. Yet, same code works well when generated with ```nextpnr-himbaechel```. I test it on simple blinky example in many different ways. It used to work fine some five months ago, and stopped working after I upgraded to latest source tree. Any ideas what's happening ? Is ```nextpnr-gowin``` deprecated and no longer maintained ?

PS: My device is TangNano-9K (GW1N-9C). I can provide a tarball with a test project with generated JASON and FS for analisys if anyone cares.

yrabbit, 2024-08-26:
You're right on almost all counts - in fact nextpnr-gowin doesn't get updates and there's little reason to use it. However, we do check the performance of examples using it when we make changes. 
You can try master-apicula and master-nextpnr, the standard examples in the package work for these versions:

https://github.com/user-attachments/assets/b0b95686-196e-43f0-82ae-529aa742f015

Attach the project files - I'm interested in what you're doing there. But there is a high probability that the next PR in nextpnr-gowin will output a warning about its obsolescence.

pointcheck, 2024-08-27:
Thank you for your response @yrabbit, I'll stick with ```nextpnr-himbaechel```.

In case you are still interested, I'm attaching a simple blinky project that does not work when built with nextpnr-gowin.

Just out of curiosity, what the word "himbaechel" means ? Is it someone's name ? :)

[tangnano9k_led-broken-nextpnr-gowin.zip](https://github.com/user-attachments/files/16758116/tangnano9k_led-broken-nextpnr-gowin.zip)

yrabbit, 2024-08-27:
It's a joke. It is intended for large architectures -> Big Arks -> Himbaechel

https://de.wikipedia.org/wiki/Himb%C3%A4chel-Viadukt

pointcheck, 2024-08-27:
Yes, exactly this bistream does not work on my tang-nano-9k boards (I have two of them ordered from famous chinese website). The thing is it used to work fine some time ago. Rebuilding with ```nextpnr-himbaechel``` makes it working again.

yrabbit, 2024-08-27:
Well, it means that it will remain a mystery because on my board this .FS functions as you see, I have nothing to work on. I think you can close Issue :)

pointcheck, 2024-08-27:
Here's a video from my side showing this same FS stalls. Just for the history.

https://github.com/user-attachments/assets/35b9066c-5cc0-48cc-bca6-e5b5af49e186

closed 2024-08-27 by pointcheck