YosysHQ/nextpnr #1362: 'Device' object has no attribute 'wire_delay'

TimSchimansky, 2024-09-09:
Hello.

I want to get my feet wet with FPGAs via Amaranth-HDL. The toolchain needs `NEXTPNR_HIMBAECHEL`.

I get this error on the `make -j$(nproc)` step all the time. I have installed `apicula` via `yay` on arch. I do not know, what else to do. I cannot install `apycula` via `yay -S python-apycula`, as it seems to not bee in that repo. I have also tried to reduce the number of FPGAs, as I am really only interested in the `GW1N-9`, but same thing.

(System: Endeavour-OS; x86; all newest updates)

```
[ 90%] Built target chipdb-himbaechel-xilinx
[ 90%] Built target chipdb-himbaechel-example
[ 90%] Built target bbasm
[ 90%] Generating ../../../share/himbaechel/gowin/chipdb-GW1N-9.bin
[ 90%] Built target nextpnr-himbaechel
Traceback (most recent call last):
  File "/home/tim/Tools/nextpnr/himbaechel/uarch/gowin/gowin_arch_gen.py", line 1386, in <module>
    main()
  File "/home/tim/Tools/nextpnr/himbaechel/uarch/gowin/gowin_arch_gen.py", line 1363, in main
    create_tiletype(create_corner_tiletype, ch, db, x, y, ttyp)
  File "/home/tim/Tools/nextpnr/himbaechel/uarch/gowin/gowin_arch_gen.py", line 591, in create_tiletype
    create_hclk_switch_matrix(tt, db, x, y)
  File "/home/tim/Tools/nextpnr/himbaechel/uarch/gowin/gowin_arch_gen.py", line 380, in create_hclk_switch_matrix
    tt.create_pip(src, dst, get_tm_class(db, "X01")) # XXX
                            ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/tim/Tools/nextpnr/himbaechel/uarch/gowin/gowin_arch_gen.py", line 238, in get_tm_class
    assert wire in db.wire_delay, f"Unknown timing class for {wire}"
                   ^^^^^^^^^^^^^
AttributeError: 'Device' object has no attribute 'wire_delay'
make[2]: *** [himbaechel/uarch/gowin/CMakeFiles/chipdb-himbaechel-gowin.dir/build.make:75: share/himbaechel/gowin/chipdb-GW1N-9.bin] Fehler 1
make[1]: *** [CMakeFiles/Makefile2:259: himbaechel/uarch/gowin/CMakeFiles/chipdb-himbaechel-gowin.dir/all] Fehler 2
make: *** [Makefile:136: all] Fehler 2
```

yrabbit, 2024-09-09:
What happens if you uninstall (I don't know how Arch packages are uninstalled) the installed apicula in this way and instead do a 

`` shell
pip install apicula
```

I just want to make sure you are using Release 0.13

TimSchimansky, 2024-09-10:
arch does not seem to want to leave pip in control and instead points to pacman/yay with a `python-` prefix:

```
╰─ pip install apicula
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try 'pacman -S
    python-xyz', where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Arch-packaged Python package,
    create a virtual environment using 'python -m venv path/to/venv'.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip.
    
    If you wish to install a non-Arch packaged Python application,
    it may be easiest to use 'pipx install xyz', which will manage a
    virtual environment for you. Make sure you have python-pipx
    installed via pacman.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
```

`pipx` does not change anything. 

If there is a possibility to use virtual environments, I would be down to try that. But I am unsure on how to use the specific venv for the compilation process...

yrabbit, 2024-09-10:
Unfortunately having DragonFlyBSD as my primary machine I have little experience with Window, Linux and Arch-linux in particular. The right thing to do would be to figure out what version yay is installing and fix it.

But we can try to fix it manually.

Make sure that apicula is installed only once - that is, repeat the original situation when you first found the bug. 

Compiling nextpnr-himbaechel uses chip bases from apicula, i.e. files with the .pickle extension.

Find a directory with such files (GW1N-9C.pickle etc), there is only one such directory (if there is not, you know we have other problems - several versions of apicula).

go to this directory and run python.
```
rabbit@fly ~/src/apicula/apycula% python
Python 3.9.14 (main, Jan  8 2023, 21:19:18)
[GCC 8.3 [DragonFly] Release/2019-02-22] on dragonfly6
Type "help", "copyright", "credits" or "license" for more information.
>>> import pickle
>>> import gzip
>>> with gzip.open('GW1N-9C.pickle', 'rb') as f:
...     db = pickle.load(f)
...
>>> db.wire_delay["X01"]
'X0'
>>>
```

if an error occurs, then we are in the right directory and can download ready databases releasef 0.13 - these are the files 
*-chipdb at the bottom of the page. You need all of them

https://github.com/YosysHQ/apicula/actions/runs/10479410531

download, unpack and overwrite them instead of the old chip database files. You can check again in python that now db.wire_delay does not cause an error.

this should help when comilating nextpnr-himbaechel.

TimSchimansky, 2024-09-11:
Ok seemlngly the build system does want to get apycula from the base python installation even when APYCULA_INSTALL_PREFIX environment varible is set to the venv, where it is definetly installed. Circumvented it by force installing apycula to the base env and removing it afterwards. But not very clean...

closed 2024-09-11 by TimSchimansky