YosysHQ/nextpnr #1364: Himbaechel (xilinx) : Placer crashes with out_of_range exception related to dict of cells

marzoul, 2024-09-10:
Hi,

When experimenting with a complete design for board Zybo (Zynq + accelerator), I get a crash in the placer.
Here is the stacktrace:

```
Info: Placed 1 cells based on constraints.
terminate called after throwing an instance of 'std::out_of_range'
  what():  dict::at()

Program received signal SIGABRT, Aborted.
0x00007ffff79b63f4 in ?? () from /usr/lib/libc.so.6
(gdb) bt
#0  0x00007ffff79b63f4 in ?? () from /usr/lib/libc.so.6
#1  0x00007ffff795d120 in raise () from /usr/lib/libc.so.6
#2  0x00007ffff79444c3 in abort () from /usr/lib/libc.so.6
#3  0x00007ffff7c97b2c in __gnu_cxx::__verbose_terminate_handler () at /usr/src/debug/gcc/gcc/libstdc++-v3/libsupc++/vterminate.cc:95
#4  0x00007ffff7cadffa in __cxxabiv1::__terminate (handler=<optimized out>) at /usr/src/debug/gcc/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:48
#5  0x00007ffff7c9752a in std::terminate () at /usr/src/debug/gcc/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:58
#6  0x00007ffff7cae2b6 in __cxxabiv1::__cxa_throw (obj=<optimized out>, tinfo=0x7ffff7e76dc0 <typeinfo for std::out_of_range>, dest=0x7ffff7cc5810 <std::out_of_range::~out_of_range()>) at /usr/src/debug/gcc/gcc/libstdc++-v3/libsupc++/eh_throw.cc:98
#7  0x000055555556f400 in nextpnr_himbaechel::dict<nextpnr_himbaechel::IdString, nextpnr_himbaechel::HeAPPlacer::CellLocation, nextpnr_himbaechel::hash_ops<nextpnr_himbaechel::IdString> >::at (this=<optimized out>, key=...)
    at /usr/src/debug/nextpnr-himbaechel-git/nextpnr-himbaechel/common/kernel/hashlib.h:598
#8  0x0000555555606bfa in nextpnr_himbaechel::HeAPPlacer::total_hpwl (this=0x7fffffffab90) at /usr/src/debug/nextpnr-himbaechel-git/nextpnr-himbaechel/common/place/placer_heap.cc:823
#9  0x000055555564ad01 in nextpnr_himbaechel::HeAPPlacer::place (this=0x7fffffffab90) at /usr/src/debug/nextpnr-himbaechel-git/nextpnr-himbaechel/common/place/placer_heap.cc:161
#10 nextpnr_himbaechel::placer_heap (ctx=0x5555557b32c0, cfg=...) at /usr/src/debug/nextpnr-himbaechel-git/nextpnr-himbaechel/common/place/placer_heap.cc:1810
#11 nextpnr_himbaechel::Arch::place (this=0x5555557b32c0) at /usr/src/debug/nextpnr-himbaechel-git/nextpnr-himbaechel/himbaechel/arch.cc:217
#12 0x00005555556ff7f2 in nextpnr_himbaechel::CommandHandler::executeMain(std::unique_ptr<nextpnr_himbaechel::Context, std::default_delete<nextpnr_himbaechel::Context> >) [clone .constprop.0] (this=this@entry=0x7fffffffc5a0, 
    ctx=std::unique_ptr<nextpnr_himbaechel::Context> = {...}) at /usr/src/debug/nextpnr-himbaechel-git/nextpnr-himbaechel/common/kernel/command.cc:674
#13 0x0000555555591823 in nextpnr_himbaechel::CommandHandler::exec (this=0x7fffffffc5a0) at /usr/src/debug/nextpnr-himbaechel-git/nextpnr-himbaechel/common/kernel/command.cc:759
#14 main (argc=<optimized out>, argv=<optimized out>) at /usr/src/debug/nextpnr-himbaechel-git/nextpnr-himbaechel/himbaechel/main.cc:107
(gdb) frame 8
#8  0x0000555555606bfa in nextpnr_himbaechel::HeAPPlacer::total_hpwl (this=0x7fffffffab90) at /usr/src/debug/nextpnr-himbaechel-git/nextpnr-himbaechel/common/place/placer_heap.cc:823
823	                CellLocation &usrloc = cell_locs.at(user.cell->name);
(gdb)
```

The affected code line in placer_heap.cc line 823 is the following:
`CellLocation &usrloc = cell_locs.at(user.cell->name);`

The reproducer is too large to attach (2MB archive, design of ~10k LUTs + BRAMs + DSPs).
Here is a public link : https://cloud.univ-grenoble-alpes.fr/s/xtMFMBcasZ5m3gk

Important : To pass the mapping step prior to tis crash, need to apply the fix proposed in other PR https://github.com/YosysHQ/nextpnr/pull/1363

Regards,
Adrien

gatecat, 2024-09-11:
Thanks for these detailed bug reports and including reproducers, I will get them fixed once I'm done with travels.

marzoul, 2024-09-23:
The issue no longer happens with the patch proposed in https://github.com/YosysHQ/nextpnr/pull/1363
So I will close this issue.

closed 2024-09-23 by marzoul