YosysHQ/nextpnr #1365: Himbaechel xilinx : Unable to place RAM128X1S

marzoul, 2024-09-10:
Hi,

With a simple reproducer design involving a LUTRAM implementation of a small memory, nextpnr-himbaechel stops early with en arror about no candidate BEL to implement "macro-cell" `RAM128X1S`.

This only happens when Yosys synthesizes synchronous 1-port memories larger than 64 lines.
Certainly linked to the way Yosys generates the memories.

```
Info: Placed 1 cells based on constraints.
ERROR: Unable to place cell 'user_inst.ram.0.8.genblk1.genblk1[0].genblk1.slice', no BELs remaining to implement cell type 'RAM128X1S'
```

The reproducer is a small GHDL-generated design with read-write memory from PS7 ports to reduce issues with IOs :
https://cloud.univ-grenoble-alpes.fr/s/9Tm5GpWFDZar4fK

Regards,
Adrien

gatecat, 2024-09-26:
This should hopefully be fixed by e4dfd4e6, let me know if you have a chance to try it.

marzoul, 2024-09-29:
Looks like it works. Thank you !