YosysHQ/nextpnr #1371: Unable to find legal placement for cell '$gbuf_set_nbtn_SB_LUT4_I3_O[2]_$glb_sr' after 10001 attempts

khalfella, 2024-09-23:
```
(apio) khalfella@ceto ~/projects/github.com/fpga-verilog/icestick/dkey/p8/sequencer $ apio build
[Sun Sep 22 22:30:17 2024] Processing icestick
---------------------------------------------------------------------------------------------------------------------------------------------------------------
nextpnr-ice40 --hx1k --package tq144 --json hardware.json --asc hardware.asc --pcf icestick.pcf -q
ERROR: Unable to find legal placement for cell '$gbuf_set_nbtn_SB_LUT4_I3_O[2]_$glb_sr' after 10001 attempts, check constraints and utilisation. Use `--placer-heap-cell-placement-timeout` to change the number of attempts.
0 warnings, 1 error
scons: *** [hardware.asc] Error 255
================================================================= [ ERROR ] Took 0.42 seconds =================================================================
(apio) khalfella@ceto ~/projects/github.com/fpga-verilog/icestick/dkey/p8/sequencer $
```
This is the full error message I got when I tried to build this code. nexpnr-ice40 is not happy for some reason I could not understand.
I [asked](https://canary.discord.com/channels/613131135903596547/644952279832723482) about this issue in 1bitsquared discord server and told to add `--placer-heap-cell-placement-timeout 0` to workaround the problem. However, I think it is possible there is a bug that needs to be fixed here.

The issue is 100% reproducible with the code in https://github.com/khalfella/fpga-verilog/tree/master/icestick/dkey/p8/sequencer
Check commit 6c1c698242d8371f4d3ec9e432bc58e29bc5a773 in case I push changes to the code.

gatecat, 2024-09-30:
Does apio create some kind of build folder? Do you think you could provide it (in particular the file `hardware.json` produced by Yosys) so I can reproduce this without setting up apio and with the exact same nextpnr input as you?

khalfella, 2024-10-01:
```
https://s3.amazonaws.com/pub.s3.kspace.sh/github.com/yosyshq/nextpnr/issues/1371/hardware.json
https://s3.amazonaws.com/pub.s3.kspace.sh/github.com/yosyshq/nextpnr/issues/1371/icestick.pcf
```
apio does not create a build folder in this case. I uploaded both `hardware.json` and `icestick.pcf` file so you can reproduce the issue by calling `nextpnr-ice40` without the need to use apio.

gatecat, 2024-10-02:
That's perfect, thanks for uploading those. It indeed found a lurking bug in the placer, see https://github.com/YosysHQ/nextpnr/pull/1378

PR #1378 (2024-10-02): heap: Fix ripup criterea
Fixes #1371 (and another drive-by fix to the ripup code to prevent ripup of locked bels I spotted at the same time...)
closed 2024-10-02 by gatecat