YosysHQ/nextpnr #1377: [Himbaechel][XILINX] missing timing file.

cbalint13, 2024-10-02:
Hi !

With the new xilinx changes (HEAD) using latest prjxray & prjxray-db leads to build error.

Seems that ```kintex7/timings/slicem.sdf``` was never present in the public [prjxray-db](https://github.com/f4pga/prjxray-db/tree/master/kintex7), is there a way to bring it there (without fuzzing form scratch) ?

Or can this be skipped by nextpnr , given latest changes ?

---

(x) I am compiling with flags:
```
       -DHIMBAECHEL_XILINX_DEVICES="xc7a100t;xc7a200t;xc7a50t;xc7k70t;xc7s50;xc7z010;xc7z020" \
       -DHIMBAECHEL_PRJXRAY_DB="/usr/share/xray/database/" \
```

(x) The error message:

```
Traceback (most recent call last):
  File "/home/cbalint/rpmbuild/BUILD/nextpnr-0.7-build/nextpnr/himbaechel/uarch/xilinx/gen/xilinx_gen.py", line 389, in <module>
    main()
  File "/home/cbalint/rpmbuild/BUILD/nextpnr-0.7-build/nextpnr/himbaechel/uarch/xilinx/gen/xilinx_gen.py", line 370, in main
    slicem_sdf = parse_sdf.parse_sdf_file(path.join(xraydb_root, "timings", "slicem.sdf"))
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/cbalint/rpmbuild/BUILD/nextpnr-0.7-build/nextpnr/himbaechel/uarch/xilinx/gen/parse_sdf.py", line 101, in parse_sdf_file
    sdata = parse_sexpr_file(filename)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/cbalint/rpmbuild/BUILD/nextpnr-0.7-build/nextpnr/himbaechel/uarch/xilinx/gen/parse_sdf.py", line 86, in parse_sexpr_file
    with open(filename, 'r') as f:
         ^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/usr/share/xray/database//kintex7/timings/slicem.sdf'
```

gatecat, 2024-10-02:
The timing is very rough and ready at the moment (due to fundamental limitations in xray), just copying the files from artix7 is fine (don't rely on the Fmax from nextpnr for now anyway)

cbalint13, 2024-10-02:
> The timing is very rough and ready at the moment (due to fundamental limitations in xray), just copying the files from artix7 is fine (don't rely on the Fmax from nextpnr for now anyway)

Thank you @gatecat !

Will do it so, I close this.

closed 2024-10-02 by cbalint13