YosysHQ/nextpnr #1392: hx4k logic and RAM appear identical to hx8k

aslak3, 2024-11-08:
I've had good success with the ice40up5k and have completed two projects with it, but for my latest project I require an FPGA with more pins, and not wanting to look at BGA construction, I've settled on a iCE40HX4K-TQ144. However I've run into a problem: building a trivial blinky with --hx4k results in seemingly the wrong amount of LC and EBR being available.

Here's 3 builds of the same verilog with different nexpnr-ice40 options set:

... --freq 8 --hx1k --package tq144 --json ice40hx-test.json ...

```
Info:            ICESTORM_LC:      28/   1280     2%
Info:           ICESTORM_RAM:       0/     16     0%
Info:           ICESTORM_PLL:       0/      1     0%
```

... --freq 8 --hx4k --package tq144 --json ice40hx-test.json ...

```
Info:            ICESTORM_LC:      28/   7680     0%
Info:           ICESTORM_RAM:       0/     32     0%
Info:           ICESTORM_PLL:       0/      2     0%
```

... --freq 8 --hx8k --package ct256 --json ice40hx-test.json ...

```
Info:            ICESTORM_LC:      28/   7680     0%
Info:           ICESTORM_RAM:       0/     32     0%
Info:           ICESTORM_PLL:       0/      2     0%
```

I'm unsure where the attributes for a particular FPGA are held, within this repo or within the icestorm repo.

Am using the latest code as of today for this repo, yosys and icestorm.

gatecat, 2024-11-08:
Sssh, don't tell Lattice :P

aslak3, 2024-11-08:
Please humour me, as I'm confused. Are you saying that LC and RAM count is valid for the hx4k parts?

gatecat, 2024-11-08:
Yes, that's correct, the differences are just in packaging and marking.