YosysHQ/nextpnr #1404: Potential Issue with Cell Placement in nextpnr-ice40 Optimization

WindFrank, 2024-12-09:
Hi nextpnr team,

I am a software tester responsible for identifying bugs related to P&R (Placement & Routing) optimization. During my research on nextpnr, I came across a potential risk in the software, specifically regarding cell placement. It appears that a cell may be placed at an unreasonable location, far from other cells with which it has direct connections.

I have analyzed two P&R solutions generated by nextpnr-ice40, both designed for the hx8k arch. These two solutions were created from separate netlists, and they do not share any common devices or routing resources. In theory, when these two netlists are combined and placed and routed together by nextpnr-ice40 on the hx8k, the final solution should maintain the same relative placement. At a minimum, the combined solution should have similar performance metrics—such as delay, wire length, and slack—as the two individual solutions produced separately by nextpnr.

To test this, I merged two netlist files (`condition_program_first259.json` and `condition_program_first2931.json`) into a combined netlist file (`condition_program_first259_and_condition_program_first2931.json`). I also created a combined .pcf file by merging the two original .pcf files. After running nextpnr-ice40 with this combined file:

```
nextpnr-ice40 --hx8k --json condition_program_first259_and_condition_program_first2931.json --pcf \ condition_program_first259_and_condition_program_first2931.pcf \
--asc condition_program_first259_and_condition_program_first2931.asc \
--detailed-timing-report \
--report condition_program_first259_and_condition_program_first2931_timing.json \
--timing-allow-fail \
--write condition_program_first259_and_condition_program_first2931_result.json \
--log condition_program_first259_and_condition_program_first2931_nextpnr.log \
--freq 1000
```

I observed an issue in the generated timing report (`condition_program_first259_and_condition_program_first2931_timing.json`), where the delay for one of the paths was **4.4 ns**—compared to **7.2 ns** in the `condition_program_first259_timing.json` report.

Upon further investigation, I found that the cell in question was placed far away from its related cell in the combined netlist solution (`condition_program_first259_and_condition_program_first2931_result.json`), compared to its placement in the individual solution (`condition_program_first259_result.json`):

• `condition_program_first259_result.json`:
From X27/Y0 to X28/Y30 to X28/Y6
• `condition_program_first259_and_condition_program_first2931_result.json`:
From X27/Y0 to X24/Y5 to X26/Y5

The large difference in placement locations (**X28/Y30** vs. **X24/Y5**) seems to have contributed to the increase in delay. I would like to know if such a placement difference is acceptable in nextpnr-ice40 or if this might indicate a potential bug that caused the 2.8 ns delay disparity. Additionally, could this phenomenon provide insights for optimizing the placement and routing algorithms in nextpnr?

I am committed to identifying potential bugs in EDA optimization tools, and I would greatly appreciate your insights on this issue.
Thank you for your time, and I look forward to your response.

Best regards

Nextpnr version: nextpnr-0.7
OS: ubuntu-20.04
Compile version: gcc 9.4.0
Test cases: [test_cases_Potential_Issue_with_Cell_Placement_in_nextpnr-ice40_Optimization.zip](https://github.com/user-attachments/files/18059916/test_cases_Potential_Issue_with_Cell_Placement_in_nextpnr-ice40_Optimization.zip)

nextpnr.log:
• `condition_program_first259_nextpnr.log`:
```
Info: Critical path report for cross-domain path '<async>' -> 'posedge h$SB_IO_IN_$glb_clk':
Info: curr total
Info:  0.0  0.0  Source xO4$sb_io.D_IN_0
Info:  3.1  3.1    Net xO4$SB_IO_IN (27,0) -> (28,30)
Info:                Sink $abc$1345$auto$blifparse.cc:492:parse_blif$1348_LC.I0
Info:                Defined in:
Info:                  /home/frank/Desktop/Software/vscode/Vgenerator-release-1.0.0-20240812/HDLProgramSets/npnr_condition_program_first/condition_program_first259.v:12
Info:  0.4  3.6  Source $abc$1345$auto$blifparse.cc:492:parse_blif$1348_LC.O
Info:  0.6  4.1    Net $abc$1345$auto$alumacc.cc:474:replace_alu$96.BB[0] (28,30) -> (28,30)
Info:                Sink $auto$alumacc.cc:474:replace_alu$96.slice[0].adder_LC.I2
Info:                Defined in:
Info:                  /home/frank/Desktop/Software/vscode/Vgenerator-release-1.0.0-20240812/HDLProgramSets/npnr_condition_program_first/condition_program_first259.v:47
Info:                  /usr/bin/../share/yosys/ice40/arith_map.v:42
Info:  0.4  4.5  Source $auto$alumacc.cc:474:replace_alu$96.slice[0].adder_LC.O
Info:  2.2  6.7    Net $abc$1345$sub$/home/frank/Desktop/Software/vscode/Vgenerator-release-1.0.0-20240812/HDLProgramSets/npnr_condition_program_first/condition_program_first259.v:47$40_Y[0] (28,30) -> (28,6)
Info:                Sink $abc$1345$auto$blifparse.cc:492:parse_blif$1346_LC.I1
Info:                Defined in:
Info:                  /home/frank/Desktop/Software/vscode/Vgenerator-release-1.0.0-20240812/HDLProgramSets/npnr_condition_program_first/condition_program_first259.v:47
Info:  0.4  7.2  Setup $abc$1345$auto$blifparse.cc:492:parse_blif$1346_LC.I1
Info: 1.2 ns logic, 5.9 ns routing
```
• `condition_program_first259_and_condition_program_first2931_nextpnr.log`:
```
Info: Critical path report for cross-domain path '<async>' -> 'posedge h$SB_IO_IN_$glb_clk':
Info: curr total
Info:  0.0  0.0  Source xO4$sb_io.D_IN_0
Info:  1.7  1.7    Net xO4$SB_IO_IN (27,0) -> (24,5)
Info:                Sink $abc$1345$auto$blifparse.cc:492:parse_blif$1348_LC.I0
Info:                Defined in:
Info:                  /home/frank/Desktop/Software/vscode/Vgenerator-release-1.0.0-20240812/HDLProgramSets/npnr_condition_program_first/condition_program_first259.v:12
Info:  0.4  2.1  Source $abc$1345$auto$blifparse.cc:492:parse_blif$1348_LC.O
Info:  0.6  2.7    Net $abc$1345$auto$alumacc.cc:474:replace_alu$96.BB[0] (24,5) -> (24,5)
Info:                Sink $auto$alumacc.cc:474:replace_alu$96.slice[0].adder_LC.I2
Info:                Defined in:
Info:                  /home/frank/Desktop/Software/vscode/Vgenerator-release-1.0.0-20240812/HDLProgramSets/npnr_condition_program_first/condition_program_first259.v:47
Info:                  /usr/bin/../share/yosys/ice40/arith_map.v:42
Info:  0.4  3.1  Source $auto$alumacc.cc:474:replace_alu$96.slice[0].adder_LC.O
Info:  0.9  4.0    Net $abc$1345$sub$/home/frank/Desktop/Software/vscode/Vgenerator-release-1.0.0-20240812/HDLProgramSets/npnr_condition_program_first/condition_program_first259.v:47$40_Y[0] (24,5) -> (26,5)
Info:                Sink $abc$1345$auto$blifparse.cc:492:parse_blif$1346_LC.I1
Info:                Defined in:
Info:                  /home/frank/Desktop/Software/vscode/Vgenerator-release-1.0.0-20240812/HDLProgramSets/npnr_condition_program_first/condition_program_first259.v:47
Info:  0.4  4.4  Setup $abc$1345$auto$blifparse.cc:492:parse_blif$1346_LC.I1
Info: 1.2 ns logic, 3.1 ns routing
```

P&R result:
• `condition_program_first259_result.json`:
[image]

• `condition_program_first259_and_condition_program_first2931_result.json`:
[image]

Ravenslofty, 2024-12-09:
> In theory, when these two netlists are combined and placed and routed together by nextpnr-ice40 on the hx8k, the final solution should maintain the same relative placement. At a minimum, the [...]

The problem with this theory is that it isn't true: nextpnr is deterministic but chaotic, and so a change in the netlist means the output may be entirely different. I consider this a feature, not a bug, as with `--seed N` we allow the user to permute the placement to explore the design space, rather than producing the same result each time.

whitequark, 2024-12-10:
> The problem with this theory is that it isn't true: nextpnr is deterministic but chaotic, and so a change in the netlist means the output may be entirely different. I consider this a feature, not a [...]

I disagree. Sure, small changes in the input (such as the seed) can reasonably cause large changes in the specifics of the output (such as the placement of individual cells), but since nextpnr is timing-driven, the aggregate properties of the output (such as the timing delay) should not be significantly worse than monotonic when combining two inputs of the kind that are described.

What we have here is...

> I observed an issue in the generated timing report (`condition_program_first259_and_condition_program_first2931_timing.json`), where the delay for one of the paths was **4.4 ns**—compared to **7.2 ns** in the `condition_program_first259_timing.json` report.

... an almost halving of the timing delay despite the doubling of the size of the netlist, which is the inverse of the expected result, and is far too large to be acceptable as normal variation in the aggregate properties of the output due to small variations in the output.

@WindFrank, can you reproduce this if you set the target frequency not to 1000 MHz but to something more achievable (e.g. 250 MHz or 125 MHz)?

whitequark, 2024-12-10:
@WindFrank Also, I've looked at your ["Bugs found by METPR"](https://github.com/WindFrank/METPR/blob/main/Bugs_found_by_METPR.pdf) page and these are absolutely the type of bugs I would be interested in seeing more of.

rowanG077, 2024-12-10:
The paths reported are asynchronous paths which the timing based analyzer ignores (== sets criticality to zero). So they are not optimized.

whitequark, 2024-12-10:
> But honestly if you really want to have consistent timing here you need to use IO registers.

As someone who has spent a *lot* of effort getting these to work in Amaranth, I am not so sure after all this effort: it turns out that many toolchains either make it really difficult to reliably pack a flop into an FF, or in some cases it's next to impossible (it's Altera of course).

I think nextpnr should automatically pack a flop into or next to an IOB as default behavior, and offer an attribute that makes the case of inability to pack it a hard error. It is perfectly placed to do so, and would make it considerably easier for code to be portable (if it doesn't use DDR flops).

rowanG077, 2024-12-10:
I agree,, I always resort to manually instantiating the relevant primitive... It would be much nicer to have reliable automatic packing + attribute to enforce a check. 

But that is not the issue here. the input port is directly wired to combinational logic so it's impossible for nextpnr to do anything.

whitequark, 2024-12-10:
> But that's not the issue here. the input port is directly wired to combinational logic so it's impossible for nextpnr to do anything.

Yes, you're right, I didn't notice that.

WindFrank, 2024-12-10:
> The problem with this theory is that it isn't true: nextpnr is deterministic but chaotic, and so a change in the netlist means the output may be entirely different. I consider this a feature, not a bug, as with we allow the user to permute the placement to explore the design space, rather than producing the same result each time.`--seed N`

I’m really excited to receive your reply — thank you! This is my first time receiving suggestions from an open-source EDA project team. :)
Actually, the theory I presented isn't entirely accurate. Not only nextpnr, but also most other P&R tools, including VTR and commercial software, may encounter similar issues. However, I’ve found that a P&R tool has a high probability of harboring a bug if a significant disparity (e.g., a large delay difference or a slack change from positive to negative) is observed between the solutions generated individually and in combination. I previously tested an industrial EDA project with a LOC of 400,000, 80,000 of which were related to P&R. By testing scenarios like these, I successfully identified around three types of bugs related to the optimization module across five versions of the project. I also found a potential bug related to clustering in VTR ([VTR_issue_2775](https://github.com/verilog-to-routing/vtr-verilog-to-routing/issues/2775)). Therefore, although the theory is not perfect, I believe it could be useful for nextpnr as well.
I would like to know if the significant inconsistency found in these cases is meaningful for nextpnr. Specifically, the cell on the critical path is placed far from the preferred solution (X28/Y30 vs. X24/Y5). The distance between these two locations, which is nearly the width of the hx8k, is quite large. Additionally, I realize that the random seed plays a crucial role in the experiment. To further investigate, I ran the tests with ten different random seeds. In these experiments, 8 out of 10 solutions showed a high delay in the critical paths, with the cells placed around location (28,30), resulting in a "U"-shaped routing path.

|random seed | single 259                             | combined 259_and_2931 | single 259 | combined 259_and_2931 |
| 48832737     | (27,0)->**(28,29)**->(28,9)   | (27,0)->(23,5)->(27,5)      | **7.0**       | 4.7                                    |
| 22949574     | (27,0)->(28,11)->(28,7)         | (27,0)->(24,5)->(24,5) | 4.8 | 4.0 |
| 9204802       | (27,0)->**(28,29)**->(28,5)   | (27,0)->(20,5)->(27,5) | **7.0** | 5.0 |
| 66736284     | (27,0)->**(28,29)**->(28,5)   | (27,0)->(24,5)->(27,5) | **7.0** | 4.4 |
| 74602019     | (27,0)->**(28,29)**->(28,5)   | (27,0)->(19,5)->(26,5) | **7.0** | 5.3 |
| 81763728     | (27,0)->**(28,29)**->(28,10) | (27,0)->(19,5)->(24,5) | **7.0** | 5.0 |
| 1652893       | (27,0)->**(28,29)**->(28,10) | (27,0)->(24,5)->(27,5) | **6.9** | 4.4 |
| 38726020     | (27,0)->**(28,31)**->(28,5) | (27,0)->(18,5)->(26,5) | **6.8** | 5.3 |
| 58373801     | (27,0)->(28,14)->(28,14) | (27,0)->(23,5)->(26,5) | 4.8 | 4.7 |
| 11109302     | (27,0)->**(28,29)**->(28,6) | (27,0)->(23,5)->(27,5) | **7.0** | 4.7 |

Is it expected behavior for nextpnr’s optimization module to generate inferior solutions most of the time when varying the random seed in this case? In other words, could these results provide a potential direction for optimizing the algorithms used in nextpnr?

WindFrank, 2024-12-10:
> @WindFrank, can you reproduce this if you set the target frequency not to 1000 MHz but to something more achievable (e.g. 250 MHz or 125 MHz)?

@whitequark Yes, the results of the solutions generated with a target frequency of 125 MHz are the same as those generated with a target frequency of 1000 MHz.

WindFrank, 2024-12-10:
> @WindFrank Also, I've looked at your ["Bugs found by METPR"](https://github.com/WindFrank/METPR/blob/main/Bugs_found_by_METPR.pdf) page and these are absolutely the type of bugs I would be interested in seeing more of discovered.

Thank you for your affirmation! I will continue to update my tool for EDA testing and the bugs it uncovered！

rowanG077, 2024-12-10:
> I would like to know if the significant inconsistency found in these cases is meaningful for nextpnr. Specifically, the cell on the critical path is placed far from the preferred solution (X28/Y30 [...]

Currently I would not say it's meaningless because are asynchronous are unconstrained. So there is no pressure for the placer and router to optimize that path. The fmax input also doesn't factor into it since it's not a synchronous path. What would be meaningful if you can replicate this finding with a startpoint and an endpoint that are both FFs on the same clock domain.

I agree with @whitequark that with the tiny netlists in question where there are a ton of resources still available in the fabric this result should be considered a bug if fmax is not reached.

WindFrank, 2024-12-11:
> What would be meaningful is if you can replicate this finding with a startpoint and an endpoint that are both FFs on the same clock domain. I agree with @whitequark that with the tiny netlists in question where there are a ton of resources still available in the fabric this result should be considered a bug if fmax is not reached.

Thank you for your suggestion!  I will test the optimization module with additional synchronous netlists.

closed 2024-12-11 by WindFrank