YosysHQ/nextpnr #1411: ERROR: Unconstrained IO

spth, 2024-12-26:
I don't know much about synthesis or P&R tools - I'm just a user. And while I know a bit of Verilog, I have no idea about the json.

I have a design that I can synthesize with yosys without -noflatten, then route with nextpnr. But if I synthesize with yosys with -noflatten, routing with nextpnr fails. So I guess there is a problem in either yosys or nextpnr, but don't really know. Since I get the error message from nextpnr, I'm opening the ticket here. I used nextpnr from the master branch from today.

[tangnano9k2_synth.json.gz](https://github.com/user-attachments/files/18255498/tangnano9k2_synth.json.gz)

```
/home/philipp/nextpnr/nextpnr-himbaechel --json tests/cblink/tangnano9k2_synth.json --write tests/cblink/tangnano9k2.json --device GW1NR-LV9QN88PC6/I5 --vopt family=GW1N-9C --vopt cst=tangnano9k.cst
Info: Using uarch 'gowin' for device 'GW1NR-LV9QN88PC6/I5'

Info: Cell O_psram_ck[0] not found
Info: Cell O_psram_ck[1] not found
Info: Cell O_psram_ck_n[0] not found
Info: Cell O_psram_ck_n[1] not found
Info: Cell O_psram_cs_n[0] not found
Info: Cell O_psram_cs_n[1] not found
Info: Cell O_psram_reset_n[0] not found
Info: Cell O_psram_reset_n[1] not found
Info: Cell IO_psram_dq[0] not found
Info: Cell IO_psram_dq[1] not found
Info: Cell IO_psram_dq[2] not found
Info: Cell IO_psram_dq[3] not found
Info: Cell IO_psram_dq[4] not found
Info: Cell IO_psram_dq[5] not found
Info: Cell IO_psram_dq[6] not found
Info: Cell IO_psram_dq[7] not found
Info: Cell IO_psram_dq[8] not found
Info: Cell IO_psram_dq[9] not found
Info: Cell IO_psram_dq[10] not found
Info: Cell IO_psram_dq[11] not found
Info: Cell IO_psram_dq[12] not found
Info: Cell IO_psram_dq[13] not found
Info: Cell IO_psram_dq[14] not found
Info: Cell IO_psram_dq[15] not found
Info: Cell IO_psram_rwds[0] not found
Info: Cell IO_psram_rwds[1] not found
Info: Reading constraints...
Info: Create constant nets...
Info: Modify LUTs...
Info: Pack IOBs...
ERROR: Unconstrained IO:system.memory.write_en_odd_IBUF_I
0 warnings, 1 error
```

yrabbit, 2024-12-27:
That's very interesting.
Would you add verilog and cst files?

jlndch, 2025-01-26:
did you set -D LEDS_NR=6 for when you run yosys? 

> I don't know much about synthesis or P&R tools - I'm just a user. And while I know a bit of Verilog, I have no idea about the json.
> 
> I have a design that I can synthesize with yosys without -noflatten, then route with nextpnr. But if I synthesize with yosys with -noflatten, routing with nextpnr fails. So I guess there is a problem in either yosys or nextpnr, but don't really know. Since I get the error message from nextpnr, I'm opening the ticket here. I used nextpnr from the master branch from today.
> 
> [tangnano9k2_synth.json.gz](https://github.com/user-attachments/files/18255498/tangnano9k2_synth.json.gz)
> 
> ```
> /home/philipp/nextpnr/nextpnr-himbaechel --json tests/cblink/tangnano9k2_synth.json --write tests/cblink/tangnano9k2.json --device GW1NR-LV9QN88PC6/I5 --vopt family=GW1N-9C --vopt cst=tangnano9k.cst
> Info: Using uarch 'gowin' for device 'GW1NR-LV9QN88PC6/I5'
> 
> Info: Cell O_psram_ck[0] not found
> Info: Cell O_psram_ck[1] not found
> Info: Cell O_psram_ck_n[0] not found
> Info: Cell O_psram_ck_n[1] not found
> Info: Cell O_psram_cs_n[0] not found
> Info: Cell O_psram_cs_n[1] not found
> Info: Cell O_psram_reset_n[0] not found
> Info: Cell O_psram_reset_n[1] not found
> Info: Cell IO_psram_dq[0] not found
> Info: Cell IO_psram_dq[1] not found
> Info: Cell IO_psram_dq[2] not found
> Info: Cell IO_psram_dq[3] not found
> Info: Cell IO_psram_dq[4] not found
> Info: Cell IO_psram_dq[5] not found
> Info: Cell IO_psram_dq[6] not found
> Info: Cell IO_psram_dq[7] not found
> Info: Cell IO_psram_dq[8] not found
> Info: Cell IO_psram_dq[9] not found
> Info: Cell IO_psram_dq[10] not found
> Info: Cell IO_psram_dq[11] not found
> Info: Cell IO_psram_dq[12] not found
> Info: Cell IO_psram_dq[13] not found
> Info: Cell IO_psram_dq[14] not found
> Info: Cell IO_psram_dq[15] not found
> Info: Cell IO_psram_rwds[0] not found
> Info: Cell IO_psram_rwds[1] not found
> Info: Reading constraints...
> Info: Create constant nets...
> Info: Modify LUTs...
> Info: Pack IOBs...
> ERROR: Unconstrained IO:system.memory.write_en_odd_IBUF_I
> 0 warnings, 1 error
> ```
> 
> I encountered this issue on a Debian GNU/Linux testing system on amd64.

did you set -D LEDS_NR=6 for when you run yosys?

yrabbit, 2025-01-26:
> did you set -D LEDS_NR=6 for when you run yosys?
> 

It seems to be a matter of the -noflatten key - which results in I/O buffers at all levels of the hierarchy. 

So far I am inclined to simply disallow the use of this key.