YosysHQ/nextpnr #1412: nextpnr gown for nano tang 9k

Bernard13090, 2024-12-29:
MacOS 13.7.1
lot of error using:
cmake . -DARCH="himbaechel" -DHIMBAECHEL_GOWIN_DEVICES="all"

problem in the scripts with: **/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.9/bin/python3.9**
had replace by **python3** because my python3 is install in **'/usr/local/bin'**
cmake was better but after I had a lot of warning

after **make -j$(nproc)** lot of warning
after **sudo make install**

Install the project...
-- Install configuration: "Release"
-- Installing: /usr/local/bin/nextpnr-himbaechel
-- Installing: /usr/local/share/nextpnr/himbaechel/example
-- Installing: /usr/local/share/nextpnr/himbaechel/gowin
-- Installing: /usr/local/share/nextpnr/himbaechel/gowin/chipdb-GW1N-9.bin
-- Installing: /usr/local/share/nextpnr/himbaechel/gowin/chipdb-GW1N-9C.bin
-- Installing: /usr/local/share/nextpnr/himbaechel/gowin/chipdb-GW1NS-4.bin
-- Installing: /usr/local/share/nextpnr/himbaechel/gowin/chipdb-GW2A-18.bin
-- Installing: /usr/local/share/nextpnr/himbaechel/gowin/chipdb-GW2A-18C.bin
-- Installing: /usr/local/share/nextpnr/himbaechel/gowin/chipdb-GW1N-1.bin
-- Installing: /usr/local/share/nextpnr/himbaechel/gowin/chipdb-GW1NZ-1.bin
-- Installing: /usr/local/share/nextpnr/himbaechel/gowin/chipdb-GW1N-4.bin
-- Installing: /usr/local/share/nextpnr/himbaechel/xilinx
-- Installing: /usr/local/share/nextpnr/himbaechel/ng-ultra

when build only

**Starting PnR with NextPnR
Finished PnR
Task finished with errors exiting**

no **led.fs** create

Can man explain how to install in detail **nexpnr** on **MacOS** with **VSCODE** ?

yrabbit, 2024-12-30:
I don't know anything about MacOs, but maybe the people on the channel can answer:
https://matrix.to/#/#apicula:matrix.org

pepijndevos, 2024-12-30:
What commands are you running to build leds.fs?

Bernard13090, 2024-12-30:
Solved issue.

My issue was exactly with **oss-cad-suite** with **VSCode**

On MacOS, you need to open **terminal** and run : **sudo xattr -r -d com.apple.quarantine .**

Thanks