YosysHQ/nextpnr #1419: nextpnr build problem for gowin devices.

Kalyan6741, 2025-01-14:
I have getting the following error during the build of nextpnr.   
there seems to be an  error in the python script.

`[  0%] Built target chipdb-himbaechel-xilinx
[  1%] Building CXX object bba/CMakeFiles/bbasm.dir/main.cc.o
[  1%] Built target chipdb-himbaechel-example
[  2%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/common/kernel/archcheck.cc.o
[  3%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/common/kernel/basectx.cc.o
[  3%] Built target chipdb-himbaechel-ng-ultra
[  5%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/common/kernel/bits.cc.o
[  6%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/common/kernel/command.cc.o
[  7%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/common/kernel/context.cc.o
[  8%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/common/kernel/design_utils.cc.o
[ 10%] Linking CXX executable bbasm
[ 10%] Built target bbasm
[ 11%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/common/kernel/embed.cc.o
[ 12%] Generating ../../../share/himbaechel/gowin/chipdb-GW1N-1.bin
[ 14%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/common/kernel/handle_error.cc.o
Traceback (most recent call last):
  File "/home/kalyan/nextpnr/himbaechel/uarch/gowin/gowin_arch_gen.py", line 1470, in <module>
    main()
  File "/home/kalyan/nextpnr/himbaechel/uarch/gowin/gowin_arch_gen.py", line 1429, in main
    create_packages(ch, db)
  File "/home/kalyan/nextpnr/himbaechel/uarch/gowin/gowin_arch_gen.py", line 1227, in create_packages
    if variant in db.sip_cst and pkgname in db.sip_cst[variant]:
                  ^^^^^^^^^^
AttributeError: 'Device' object has no attribute 'sip_cst'
make[2]: *** [himbaechel/uarch/gowin/CMakeFiles/chipdb-himbaechel-gowin.dir/build.make:82: share/himbaechel/gowin/chipdb-GW1N-1.bin] Error 1
make[1]: *** [CMakeFiles/Makefile2:277: himbaechel/uarch/gowin/CMakeFiles/chipdb-himbaechel-gowin.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 15%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/common/kernel/idstring.cc.o
[ 16%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/common/kernel/idstringlist.cc.o
[ 17%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/common/kernel/log.cc.o
[ 19%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/common/kernel/nextpnr.cc.o
[ 20%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/common/kernel/nextpnr_assertions.cc.o
[ 21%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/common/kernel/nextpnr_namespaces.cc.o
[ 23%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/common/kernel/nextpnr_types.cc.o
[ 24%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/common/kernel/property.cc.o
[ 25%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/common/kernel/pybindings.cc.o
[ 26%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/common/kernel/report.cc.o
[ 28%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/common/kernel/sdc.cc.o
[ 29%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/common/kernel/sdf.cc.o
[ 30%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/common/kernel/str_ring_buffer.cc.o
[ 32%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/common/kernel/svg.cc.o
[ 33%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/common/kernel/timing.cc.o
[ 34%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/common/kernel/timing_log.cc.o
[ 35%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/common/place/detail_place_core.cc.o
[ 37%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/common/place/parallel_refine.cc.o
[ 38%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/common/place/place_common.cc.o
[ 39%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/common/place/placer1.cc.o
[ 41%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/common/place/placer_heap.cc.o
[ 42%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/common/place/placer_static.cc.o
[ 43%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/common/place/timing_opt.cc.o
[ 44%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/common/route/router1.cc.o
[ 46%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/common/route/router2.cc.o
[ 47%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/3rdparty/json11/json11.cpp.o
[ 48%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/3rdparty/oourafft/fftsg.cc.o
[ 50%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/3rdparty/oourafft/fftsg2d.cc.o
[ 51%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/json/jsonwrite.cc.o
[ 52%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/frontend/json_frontend.cc.o
[ 53%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/himbaechel/arch.cc.o
[ 55%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/himbaechel/arch_pybindings.cc.o
[ 56%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/himbaechel/himbaechel_api.cc.o
[ 57%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/himbaechel/himbaechel_helpers.cc.o
[ 58%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/himbaechel/main.cc.o
[ 60%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/himbaechel/uarch/example/example.cc.o
[ 61%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/himbaechel/uarch/gowin/cst.cc.o
[ 62%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/himbaechel/uarch/gowin/globals.cc.o
[ 64%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/himbaechel/uarch/gowin/gowin.cc.o
[ 65%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/himbaechel/uarch/gowin/gowin_utils.cc.o
[ 66%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/himbaechel/uarch/gowin/pack.cc.o
[ 67%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/himbaechel/uarch/xilinx/cells.cc.o
[ 69%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/himbaechel/uarch/xilinx/fasm.cc.o
[ 70%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/himbaechel/uarch/xilinx/pack.cc.o
[ 71%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/himbaechel/uarch/xilinx/pack_carry.cc.o
[ 73%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/himbaechel/uarch/xilinx/pack_clocking.cc.o
In file included from /home/kalyan/nextpnr/himbaechel/uarch/xilinx/fasm.cc:32:
/home/kalyan/nextpnr/common/kernel/util.h: In instantiation of ‘bool nextpnr_himbaechel::boolstr_or_default(const dict<KeyType, Property>&, const KeyType&, bool) [with KeyType = IdString]’:
/home/kalyan/nextpnr/himbaechel/uarch/xilinx/fasm.cc:1536:44:   required from here
/home/kalyan/nextpnr/common/kernel/util.h:114:9: warning: value computed is not used [-Wunused-value]
  114 |         bool(found->second.as_int64());
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[ 74%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/himbaechel/uarch/xilinx/pack_dram.cc.o
[ 75%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/himbaechel/uarch/xilinx/pack_dsp_xc7.cc.o
[ 76%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/himbaechel/uarch/xilinx/pack_io.cc.o
[ 78%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/himbaechel/uarch/xilinx/pins.cc.o
[ 79%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/himbaechel/uarch/xilinx/xdc.cc.o
[ 80%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/himbaechel/uarch/xilinx/xilinx.cc.o
[ 82%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/himbaechel/uarch/xilinx/xilinx_place.cc.o
[ 83%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/himbaechel/uarch/ng-ultra/bitstream.cc.o
[ 84%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/himbaechel/uarch/ng-ultra/cells.cc.o
[ 85%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/himbaechel/uarch/ng-ultra/csv.cc.o
[ 87%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/himbaechel/uarch/ng-ultra/location_map.cc.o
[ 88%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/himbaechel/uarch/ng-ultra/ng_ultra.cc.o
[ 89%] Building CXX object CMakeFiles/nextpnr-himbaechel.dir/himbaechel/uarch/ng-ultra/pack.cc.o
[ 91%] Linking CXX executable nextpnr-himbaechel
[ 91%] Built target nextpnr-himbaechel
make: *** [Makefile:136: all] Error 2
`

whitequark, 2025-01-14:
That seems like your Apycula version might be wrong.

Kalyan6741, 2025-01-14:
i used this 
`sudo apt install python3-apycula`

i tried `pip install apycula`  
but i got this message 

`× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.12/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
`

is there another way?

whitequark, 2025-01-14:
The error message is telling you what to do: use a virtual environment.

closed 2025-01-14 by Kalyan6741