YosysHQ/nextpnr #1420: missing GW1N-1.pickle

Kalyan6741, 2025-01-14:
im having trouble building nextpnr for Gowin

`[ 87%] Built target nextpnr-himbaechel
[ 89%] Built target bbasm
[ 89%] Built target chipdb-himbaechel-example
[ 91%] Generating ../../../share/himbaechel/gowin/chipdb-GW1N-1.bin
Traceback (most recent call last):
  File "/home/kalyan/nextpnr/himbaechel/uarch/gowin/gowin_arch_gen.py", line 1470, in <module>
    main()
  File "/home/kalyan/nextpnr/himbaechel/uarch/gowin/gowin_arch_gen.py", line 1400, in main
    with gzip.open(importlib.resources.files("apycula").joinpath(f"{device}.pickle"), 'rb') as f:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/gzip.py", line 61, in open
    binary_file = GzipFile(filename, gz_mode, compresslevel)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/gzip.py", line 192, in __init__
    fileobj = self.myfileobj = builtins.open(filename, mode or 'rb')
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/home/kalyan/apicula/venv/lib/python3.12/site-packages/apycula/GW1N-1.pickle'
make[2]: *** [himbaechel/uarch/gowin/CMakeFiles/chipdb-himbaechel-gowin.dir/build.make:82: share/himbaechel/gowin/chipdb-GW1N-1.bin] Error 1
make[1]: *** [CMakeFiles/Makefile2:277: himbaechel/uarch/gowin/CMakeFiles/chipdb-himbaechel-gowin.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
`
ive installed apycula in venv as it was recommended

whitequark, 2025-01-14:
Please use ` ``` ` to delimit your code pastes since they are otherwise hard to read. Also, please do not open a new issue for every configuration problem you encounter.

Kalyan6741, 2025-01-14:
sorry. will take care next time.

closed 2025-01-14 by Kalyan6741