YosysHQ/nextpnr #1428: Build issue related to pybind11_headers

mpbraendli, 2025-01-20:
Hello all,

running arch linux, pybind 2.13.6, python 3.13.1, and building nextpnr 0.7_149_g155adc3f from AUR using this PKGBUILD: https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=nextpnr-ice40-nightly

Linking fails because `pybind11_headers` isn't found.

Linking suceeds with the following patch: 

```patch
diff --git a/common/kernel/CMakeLists.txt b/common/kernel/CMakeLists.txt
index d813d684..cf2f2f5b 100644
--- a/common/kernel/CMakeLists.txt
+++ b/common/kernel/CMakeLists.txt
@@ -74,8 +74,3 @@ target_link_libraries(nextpnr_kernel INTERFACE
     json11
 )
 
-if (BUILD_PYTHON)
-    target_link_libraries(nextpnr_kernel INTERFACE
-        pybind11_headers
-    )
-endif()
```

I don't know in what situations this `target_link_libraries` is required, but on this system it poses a problem.

Hope this report helps,

Cheers
mpb

whitequark, 2025-01-20:
Please post the CMake log; I'd like to understand why pybind11_headers are not available.

lethalbit, 2025-01-20:
I didn't notice this open prior to opening #1430, so i'll repeat what I mentioned there and close that.

The problem is that the `pybind11_headers` leaks out onto the link line as `-lpybind11_headers`, and that's not valid when using the system pybind11, it seems to only be needed when using the bundled 3rdparty pybind11, so indiscriminately using this is problematic. 

Attempting to build with the vendored pybind11 by setting `pybind11_FOUND` to FALSE also causes a build failure to cmake being confused and ending up with a handful of redefinitions. I'll pull the actual log in a moment when I have a second, but regardless it's still busted in either case if the host system has pybind11 installed on it.

mpbraendli, 2025-01-20:
CMake gets through, it's the build itself that fails. What logs do you need exactly?

```
[1/2] Linking CXX executable nextpnr-ice40
FAILED: nextpnr-ice40 
: && /usr/bin/c++ -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions         -Wp,-D_FORTIFY_SOURCE=3 -Wformat -Werror=format-security         -fstack-clash-protection -fcf-protection         -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -Wp,-D_GLIBCXX_ASSERTIONS -g -ffile-prefix-map=/home/bram/.cache/paru/clone/nextpnr-ice40-nightly/src=/usr/src/debug/nextpnr-ice40-nightly -flto=auto -O2 -g -DNDEBUG -flto=auto -fno-fat-lto-objects -Wl,-O1 -Wl,--sort-common -Wl,--as-needed -Wl,-z,relro -Wl,-z,now          -Wl,-z,pack-relative-relocs -flto=auto   -Wl,--export-dynamic -rdynamic  -Wl,--dependency-file=CMakeFiles/nextpnr-ice40.dir/link.d CMakeFiles/chipdb-ice40.dir/ice40/chipdb/chipdb-384.cc.o CMakeFiles/chipdb-ice40.dir/ice40/chipdb/chipdb-1k.cc.o CMakeFiles/chipdb-ice40.dir/ice40/chipdb/chipdb-5k.cc.o CMakeFiles/chipdb-ice40.dir/ice40/chipdb/chipdb-u4k.cc.o CMakeFiles/chipdb-ice40.dir/ice40/chipdb/chipdb-8k.cc.o CMakeFiles/nextpnr-ice40.dir/ice40/arch.cc.o CMakeFiles/nextpnr-ice40.dir/ice40/arch_place.cc.o CMakeFiles/nextpnr-ice40.dir/ice40/arch_pybindings.cc.o CMakeFiles/nextpnr-ice40.dir/ice40/bitstream.cc.o CMakeFiles/nextpnr-ice40.dir/ice40/cells.cc.o CMakeFiles/nextpnr-ice40.dir/ice40/chains.cc.o CMakeFiles/nextpnr-ice40.dir/ice40/delay.cc.o CMakeFiles/nextpnr-ice40.dir/ice40/gfx.cc.o CMakeFiles/nextpnr-ice40.dir/ice40/main.cc.o CMakeFiles/nextpnr-ice40.dir/ice40/pack.cc.o CMakeFiles/nextpnr-ice40.dir/ice40/pcf.cc.o CMakeFiles/nextpnr-ice40.dir/common/kernel/archcheck.cc.o CMakeFiles/nextpnr-ice40.dir/common/kernel/basectx.cc.o CMakeFiles/nextpnr-ice40.dir/common/kernel/bits.cc.o CMakeFiles/nextpnr-ice40.dir/common/kernel/command.cc.o CMakeFiles/nextpnr-ice40.dir/common/kernel/context.cc.o CMakeFiles/nextpnr-ice40.dir/common/kernel/design_utils.cc.o CMakeFiles/nextpnr-ice40.dir/common/kernel/embed.cc.o CMakeFiles/nextpnr-ice40.dir/common/kernel/handle_error.cc.o CMakeFiles/nextpnr-ice40.dir/common/kernel/idstring.cc.o CMakeFiles/nextpnr-ice40.dir/common/kernel/idstringlist.cc.o CMakeFiles/nextpnr-ice40.dir/common/kernel/log.cc.o CMakeFiles/nextpnr-ice40.dir/common/kernel/nextpnr_assertions.cc.o CMakeFiles/nextpnr-ice40.dir/common/kernel/nextpnr.cc.o CMakeFiles/nextpnr-ice40.dir/common/kernel/nextpnr_namespaces.cc.o CMakeFiles/nextpnr-ice40.dir/common/kernel/nextpnr_types.cc.o CMakeFiles/nextpnr-ice40.dir/common/kernel/property.cc.o CMakeFiles/nextpnr-ice40.dir/common/kernel/pybindings.cc.o CMakeFiles/nextpnr-ice40.dir/common/kernel/report.cc.o CMakeFiles/nextpnr-ice40.dir/common/kernel/sdc.cc.o CMakeFiles/nextpnr-ice40.dir/common/kernel/sdf.cc.o CMakeFiles/nextpnr-ice40.dir/common/kernel/str_ring_buffer.cc.o CMakeFiles/nextpnr-ice40.dir/common/kernel/svg.cc.o CMakeFiles/nextpnr-ice40.dir/common/kernel/timing.cc.o CMakeFiles/nextpnr-ice40.dir/common/kernel/timing_log.cc.o CMakeFiles/nextpnr-ice40.dir/frontend/json_frontend.cc.o CMakeFiles/nextpnr-ice40.dir/json/jsonwrite.cc.o CMakeFiles/nextpnr-ice40.dir/common/place/detail_place_core.cc.o CMakeFiles/nextpnr-ice40.dir/common/place/parallel_refine.cc.o CMakeFiles/nextpnr-ice40.dir/common/place/place_common.cc.o CMakeFiles/nextpnr-ice40.dir/common/place/placer1.cc.o CMakeFiles/nextpnr-ice40.dir/common/place/placer_heap.cc.o CMakeFiles/nextpnr-ice40.dir/common/place/placer_static.cc.o CMakeFiles/nextpnr-ice40.dir/common/place/timing_opt.cc.o CMakeFiles/nextpnr-ice40.dir/common/route/router1.cc.o CMakeFiles/nextpnr-ice40.dir/common/route/router2.cc.o CMakeFiles/nextpnr-ice40.dir/3rdparty/oourafft/fftsg.cc.o CMakeFiles/nextpnr-ice40.dir/3rdparty/oourafft/fftsg2d.cc.o -o nextpnr-ice40  /usr/lib/libboost_filesystem.so.1.86.0  /usr/lib/libboost_program_options.so.1.86.0  /usr/lib/libboost_iostreams.so.1.86.0  /usr/lib/libboost_system.so.1.86.0  /usr/lib/libboost_thread.so.1.86.0  -lpthread  generated/gui/ice40/libgui_ice40.a  /usr/lib/libpython3.13.so  3rdparty/json11/libjson11.a  -lpybind11_headers  /usr/lib/libtbb.so.12.14  /usr/lib/libQt5OpenGL.so.5.15.16  generated/3rdparty/QtPropertyBrowser/src/libQtPropertyBrowser.a  /usr/lib/libQt5Widgets.so.5.15.16  /usr/lib/libQt5Gui.so.5.15.16  /usr/lib/libQt5Core.so.5.15.16 && :
/usr/bin/ld: cannot find -lpybind11_headers: No such file or directory
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

whitequark, 2025-01-20:
Huh, why are you un-vendoring pybind11?

lethalbit, 2025-01-20:
It's not being un-vendored, CMake is picking up the system pybind11 and using that over the vendored copy.

And trying to persuade it to *not* do that causes even more issues.

whitequark, 2025-01-20:
Oh got it. I will look into this!

whitequark, 2025-01-21:
It turns out that the proper fix is just two characters long: https://github.com/YosysHQ/nextpnr/pull/1432

whitequark, 2025-01-21:
I've checked that the fix works with both built-in and system pybind11 on my Debian system, and I think it should work anywhere, but if it doesn't let me know and I'll chase that down too.

lethalbit, 2025-01-22:
@whitequark how was this tested, because as of 17943a51, at least on Arch Linux it's **still** trying to link with `-lpybind11_headers`

The line used to configure/build to test this was:
```
cmake -G Ninja -DARCH=ice40 -DBUILD_PYTHON=ON -DBUILD_GUI=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_INSTALL_PREFIX=/usr
```

For reference:
pybind11: 2.13.6
cmake: 3.31.4

whitequark, 2025-01-22:
> how was this tested,

```
sudo apt install pybind11-dev; cmake -S . -B build1 -DARCH=generic && cmake --build build1
# -- Found pybind11: //include (found version "2.10.3")
sudo apt remove  pybind11-dev; cmake -S . -B build2 -DARCH=generic && cmake --build build2
# -- Could NOT find pybind11 (missing: pybind11_DIR)
```

But I see now what the issue was: I didn't build with the GUI enabled. I'll fix this in a moment.

whitequark, 2025-01-22:
https://github.com/YosysHQ/nextpnr/pull/1434

commit 43b2f385 (2025-01-21): CMake: use imported target for pybind11.

This accounts for the use of either the system or the vendored pybind11.

Fixes #1428.