YosysHQ/nextpnr #1429: himbaechel-test: SegFault, Unable to read chipdb

xiota, 2025-01-20:
```
Internal ctest changing into directory: /home/builder/build/src/build
Test project /home/builder/build/src/build
    Start 1: himbaechel-test
1/2 Test #1: himbaechel-test ..................***Exception: SegFault  0.01 sec
[==========] Running 5 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 5 tests from NGUltraLutDffTest
[ RUN      ] NGUltraLutDffTest.pack_constants
Info: Using uarch 'ng-ultra' for device 'NG-ULTRA'
ERROR: Unable to read chipdb /home/builder/build/src/build/share/himbaechel/ng-ultra/chipdb-ng-ultra.bin
unknown file: Failure
Unknown C++ exception thrown in SetUp().

    Start 2: generic-test
2/2 Test #2: generic-test .....................   Passed    0.01 sec

50% tests passed, 1 tests failed out of 2

Total Test time (real) =   0.02 sec

The following tests FAILED:
          1 - himbaechel-test (SEGFAULT)
Errors while running CTest
```

git bisect:
```
# good: [5eaa1b3f1f34595604aa5f919c0b00033509923f] Gowin. Add IODELAY. (#1398)
# bad: [284fb3e874b5755690792dcc38f513e0e4aaad2d] Updating CI to work with ubuntu-latest (#1426)
# bad: [55bd7608089bd5b9efd0866744f491a99a2ba330] ice40: Don't constrain multiple potentially-incompatible FFs to same tile (#1413)
# bad: [f01465f6280e2eae2459819597fda0fc2f46b703] Add attributes to the hierarchical cells (#1405)
# bad: [d810aac8676b9b4af0f4a1145bcfe9ac2a0b102e] Add GroupId related calls to Himbaechel API (#1399)
# bad: [5a807110de045e5d3c16d878072f54f9e69421f3] Adding NanoXplore NG-Ultra support (#1397)
# first bad commit: [5a807110de045e5d3c16d878072f54f9e69421f3] Adding NanoXplore NG-Ultra support (#1397)
```

whitequark, 2025-01-20:
Himbaechel segfaults if you run `--test` and the chipdb isn't available. This will actually be fixed by #1422 soon, since NG-Ultra (of very little interest to most people) will no longer be built by default.

xiota, 2025-01-21:
Appears to be resolved in latest commit (17943a51cb1499f0a465bc4deb9d2a2d92876730) with new CMake options.  `-DARCH="himbaechel;generic" -DHIMBAECHEL_UARCH=gowin`

closed 2025-01-21 by xiota