YosysHQ/nextpnr #1443: Commit 1623243 (for cmake) changess way to build from source.

dennowiggle, 2025-01-28:
Can someone point me to the right cmake command to use in order to build nextpnr-ice40 now. Recent commit 1623243 means the instructions on the nextpnr README are no longer work.

This was the commit
https://github.com/YosysHQ/nextpnr/commit/1623243d50b88dd4f1536784399eedb8090f74b4

These are the insructions in the README file
```cmake . -DARCH=ice40
make -j$(nproc)
sudo make install
```

whitequark, 2025-01-28:
I just merged #1441 which should answer your question. If the build fails anyway try `make -j1`; this is a known issue we're working on.

dennowiggle, 2025-01-28:
Thanks for the quick response. 

Can I ignore the two error mesages
1.  `add_subdirectory given source "tests/gui" which is not an existing directory` and 
2. `Configuring incomplete, errors occurred!` 

Full log
```-- Building with IPO
-- C++ compiler does NOT support #embed
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found version "1.74.0") found components: filesystem program_options iostreams system thread
-- Could NOT find pybind11 (missing: pybind11_DIR)
-- Using built-in pybind11
-- pybind11 v2.12.1
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found version "1.74.0") found components: program_options filesystem system
CMake Error at CMakeLists.txt:242 (add_subdirectory):
  add_subdirectory given source "tests/gui" which is not an existing
  directory.

-- Configuring architecture: ice40
-- IceStorm install prefix: /usr/local
-- icebox data directory: /usr/local/share/icebox
-- Enabled iCE40 devices: 384;1k;5k;u4k;8k
-- Configuring incomplete, errors occurred!
```

whitequark, 2025-01-28:
You did not run `git submodule update --init`.

dennowiggle, 2025-01-28:
Thanks for the help with the  latest commits to the README file. I can build nextpnr again.

closed 2025-01-28 by dennowiggle