YosysHQ/nextpnr #1449: chipdb build errors when `-Wnarrowing` is on by default

gsomlo, 2025-01-31:
On fedora 42 (currently rawhide/development, using gcc-15.0.1), I see a very large number of errors similar to the following:

```
/builddir/build/BUILD/nextpnr-1-build/nextpnr-81ccada/redhat-linux-build/ice40/chipdb-1k.bin.cc:6:1:
   error: narrowing conversion of ‘163’ from ‘int’ to ‘const char’ [-Wnarrowing]
```

These are `.cc` files auto-generated based on imported chip database files (from icestorm, in this particular case), and look something like this:

```
#include "nextpnr.h"
#include "embed.h"
NEXTPNR_NAMESPACE_BEGIN
const char chipdb_blob_1k[17888027] = {
#embed "/builddir/build/BUILD/nextpnr-1-build/nextpnr-81ccada/redhat-linux-build/ice40/chipdb-1k.bin"
};
EmbeddedFile chipdb_file_1k("ice40/chipdb-1k.bin", chipdb_blob_1k);
NEXTPNR_NAMESPACE_END
```

I'm not sufficiently spun up on c++ minutiae to come up with an immediate fix, but I noticed that forcibly adding `-Wno-narrowing` to CXXFLAGS will (for now) paper over the issue, and allow the build to complete successfully.

My question is whether using `-Wno-narrowing` is the right thing to do in the medium/long term, or whether there's something that can/should be done to address the root cause.

Thanks!

whitequark, 2025-01-31:
What if you swap `const char` to `const uint8_t`?

gsomlo, 2025-01-31:
> What if you swap `const char` to `const uint8_t`?

which file in the actual nextpnr sources generates the `chipdb-*.bin.cc` files (i.e., where's the template I'd have to touch to make that change)?

whitequark, 2025-01-31:
`bba/main.cc`

gsomlo, 2025-01-31:
yeah, `uint8_t` instead of `char` takes care of it (see also https://github.com/YosysHQ/nextpnr/pull/1450) :)

Thanks!

closed 2025-02-01 by whitequark