YosysHQ/nextpnr #1452: nextpnr-himbaechel routing fails for GW2AR-LV18QN88C/I7

Niko-the-Useless, 2025-02-10:
command im running:
nextpnr-himbaechel --json counter.json --write route.json --device GW2AR-LV18QN88C/I7 --vopt cst=pins.cst --vopt family=GW2A-18

output:
Info: Using uarch 'gowin' for device 'GW2AR-LV18QN88C/I7'
terminate called after throwing an instance of 'nextpnr_himbaechel::assertion_failure'
  what():  Assertion failure: int(ctx->idstring_idx_to_str->size()) == idx (/work/_builds/linux-x64/nextpnr-himbaechel/nextpnr/common/kernel/idstring.cc:46)
[1]    124166 IOT instruction  sudo nextpnr-himbaechel --json counter.json --write route.json --device

i have no idea what could be the problem since i never worked with FPGA. my board is tang nano 20k if that matters

gatecat, 2025-02-10:
This error usually occurs when the chip database has got out of sync with the nextpnr binary - I'd recommend trying a clean rebuild first.

Niko-the-Useless, 2025-02-10:
thanks, it fixed all the issues, before i was using nextpnr build from oss-cad-suite and i guess there is something wrong there

yrabbit, 2025-02-10:
btw if your board is Tangnano20k, the family should be GW2A-18C.

closed 2025-06-05 by Niko-the-Useless