YosysHQ/nextpnr #1454: nextpnr-himbaechel core dumps on simple gowin example design

philtomson, 2025-02-12:
I was told on IRC that nextpnr-gowin in deprecated and to use nextpnr-himbaechel. I ran the following commandline:

$ nextpnr-himbaechel  --json counter.json --write counter_pnr.json --freq 27 --device GW1NR-LV9QN88PC6/I5 --vopt family=GW1N-9C --vopt cst=tangnano9k.cst

With this result:
Info: Using uarch 'gowin' for device 'GW1NR-LV9QN88PC6/I5'
terminate called after throwing an instance of 'nextpnr_himbaechel::assertion_failure'
  what():  Assertion failure: int(ctx->idstring_idx_to_str->size()) == idx (/work/_builds/linux-x64/nextpnr-himbaechel/nextpnr/common/kernel/idstring.cc:46)
Aborted (core dumped)

This was the nextpnr-himbaechel from the 2025_02_11 oss-cad-suite release build (today's). I happened to have an older version of oss-cad-suite (2024_2_16) and tried running the exact same command line and it works (does _not_ core dump, successfully creates a counter_pnr.json file)

For the design files see:  https://github.com/lushaylabs/tangnano9k-series-examples/tree/master/counter (counter.v tangnano9k.cst)

yrabbit, 2025-02-12:
Same thing?

https://github.com/YosysHQ/nextpnr/issues/1452

xiaguangbo, 2025-03-21:
use oss-cad-suite-20250320 same problem occurred
```
yosys -D LEDS_NR=6 -D INV_BTN=1 -p "read_verilog blinky.v; synth_gowin -json blinky.json"
nextpnr-himbaechel --json blinky.json \
                   --write pnrblinky.json \
                   --device GW2AR-LV18QN88C8/I7 \
                   --vopt family=GW2A-18C \
                   --vopt cst=tangnano20k.cst
Info: Using uarch 'gowin' for device 'GW2AR-LV18QN88C8/I7'
terminate called after throwing an instance of 'nextpnr_himbaechel::assertion_failure'
  what():  Assertion failure: int(ctx->idstring_idx_to_str->size()) == idx (/work/_builds/linux-x64/nextpnr-himbaechel/nextpnr/common/kernel/idstring.cc:46)
Aborted (core dumped)
```

yrabbit, 2025-03-21:
chipdb-GW2A-18C.bin nextpnr-himbaechel files are created during nextpnr compilation, it is interesting to look at the date and time these files were created on your computer.

The essence of this crash: nextpnr uses numeric identifiers instead of strings. Roughly speaking, during compilation all strings are placed in a dictionary and further manipulations occur only with the numbers of strings in this dictionary.

A crash occurs if a string request with a number that is not in the dictionary is made. As you understand it can happen only if the dictionary and executable file do not correspond to each other, in this case the chip base was created in another place and for another executable file nextpnr.

xiaguangbo, 2025-03-21:
@yrabbit this

Is there any way to fix it?

[image]

yrabbit, 2025-03-21:
For starters, I was wondering how is it that Created is later than Modified? :) Is it Linux?

Well, about the cure - you can uninstall everything so that you don't have any chipdb-GW2A-18C.bin and nextpnr-himbaechel files OM THE WHOLE DISC. That is, nowhere at all. Then install oss-cad-suite-20250320 and make sure that you have only one instance of chipdb-GW2A-18C.bin and nextpnr-himbaechel, i.e. you have exactly installed a synchronised base-chip<->executable-file pair.

If there is an error after that, then. someone else will advise you, someone who understands oss-cad more.

xiaguangbo, 2025-03-21:
@yrabbit 
system: fedora 41 x64
use `tar -xvzf ./oss-cad-suite-linux-x64-20250320.tgz`, I tried it again, unzipped it, and it looked like this.
```
$ md5sum ./oss-cad-suite-linux-x64-20250320.tgz 
0a751188d92c7471de81506c9b5ee3d1  ./oss-cad-suite-linux-x64-20250320.tgz
```

xiaguangbo, 2025-03-21:
@yrabbit 20250321 is ok!

[image]

yrabbit, 2025-03-21:
That's great! I'm glad.

And in case anyone else comes here about a similar error, know that it's a mis-synchronisation of the chip databases and the nextnpr executable. Try another distribution of oss-cad

closed 2025-06-25 by gatecat