YosysHQ/nextpnr #1466: "Package does not have a pin named 'xx'" ice40 nextpnr

nfraly, 2025-03-15:
When trying to use pins 27 and 36 (maybe others, those are what we are running into right now) as assignments for outputs, nextpnr errors saying the package does not have these pins. We are able to use these pins in Radiant, but when we port to yosys/nextpnr, the error occurs. Has anyone run into this issue and have a workaround?

pcf:
set_io flash_clk 15
set_io flash_csn 16
set_io flash_io0 17
set_io flash_io1 14
set_io leds[0]   38
set_io leds[1]   42
set_io leds[2]   36
set_io leds[3]   43
set_io leds[4]   34
set_io leds[5]   37
set_io leds[6]   31
set_io leds[7]   32
set_io uart_rx   15
set_io uart_tx   14 
set_io -nowarn spi_mosi  27
set_io spi_clk     26
set_io spi_cs_n     25
set_io lcd_dc      23
set_io test_out  36
set_io reset_n   21

gatecat, 2025-03-15:
Which iCE40 device/package are you targeting, and what nextpnr command line?

gatecat, 2025-03-15:
Try adding `--up5k --package sg48` to the nextpnr command line - otherwise it will default to the wrong device

nfraly, 2025-03-15:
That worked, thank you! Is there way to check what it defaults to? I looked over the pinout documentation for the 3k, and the UWG30 packaging for the 5k as well, and there are other pins that work fine but didn't cause issues. I don't fully understand the control flow of nextpnr, so when I tried to look I couldn't find anything but I was mostly opening random files. Still new-ish to issues in github, should I close this?