YosysHQ/nextpnr #1468: Outdated getting started instructions for Linux?

Umoma, 2025-03-26:
I use to have no problem when building _nextpnr-0.7_ following the `README.md` instructions

[](https://github.com/YosysHQ/nextpnr#nextpnr-ice40)

which are these:

```
mkdir -p build && cd build
cmake .. -DARCH=ice40
make -j$(nproc)
sudo make install
```

However, since _nextpnr-0.8_

```
$ cmake .. -DARCH=ice40
CMake Error at CMakeLists.txt:2 (message):
  In-tree builds are not supported.  Instead, run:

  cmake .  -B build <options> && cmake --build build

-- Configuring incomplete, errors occurred!
```

**Now this works:**

```
mkdir build
cmake . -B build -DARCH=ice40
cd build
make -j$(nproc)
sudo make install
```
* May I doing something wrong? May I setting up something wrong?
* Or, is README.md outdated for version 0.8?

**System info:**
_cmake version 3.25.1_

gatecat, 2025-03-26:
These instructions should still be fine, and seem okay over here - are you sure you are in the newly created `build` subdirectory when running `cmake`?

whitequark, 2025-03-26:
This is probably happening because you have a `CMakeCache.txt` in the directory root. Try running `git clean -dxf` in the `nextpnr` checkout to clear out all of the stale build products--do note this will also remove any untracked files you may have, if you've done some local changes.

Umoma, 2025-03-26:
@whitequark many thanks your suggestion. `git clean -dxf` worked!

My explanation: most likely I've messed at first when trying to build off the top of my head instead of following the recipe from start, naturally failed, and when I tried again, this time following the instructions, failed again. Of course, deleting the `build` directory was not enough to fix it, so that's why I ended up with that modified `cmake` command!

So thanks again both for your kind support!

closed 2025-03-26 by Umoma