YosysHQ/nextpnr #1469: Differential clock pins on ECP5 EVN are not available as regular I/O pins.

mivuzu, 2025-03-30:
hi,
i've been trying to use the 200MHz clock that comes with the ECP5 EVN. According to the [datasheet (page 15)](https://www.latticesemi.com/view_document?document_id=52479), it's connected to pins Y19 and W20, which if requested on the .lpf make nextpnr error out.
i assume the pins are reserved for some other purpose or only available through a verilog module, however i couldn't find any info about this. here's the exact error message.
```
$ make layout
nextpnr-ecp5 --um5g-85k --speed 8 --package CABGA381 --json obj/hardware.json --textcfg obj/hardware.config --lpf lib/pins.lpf -q
ERROR: IO pin 'clk$tr_io' constrained to pin 'Y19', which does not exist for package 'CABGA381'.
ERROR: Packing design failed.
0 warnings, 2 errors
make: *** [Makefile:8: obj/hardware.config] Error 255
```

whitequark, 2025-03-30:
Y19/W20 are SERDES input only; you can connect them to `EXTREFCLK` but I don't believe there is any way to use the clock from the fabric.

mivuzu, 2025-03-31:
> Y19/W20 are SERDES input only; you can connect them to `EXTREFCLK` but I don't believe there is any way to use the clock from the fabric.

I see. So the balls are only connected to the inner DSP components of the chip and cannot be used by the programmable logic if I'm understanding you correctly.
I suppose the only solution in that case would be to solder a connection from the output of the oscillator to another I/O pin of board.

rowanG077, 2025-03-31:
You really do need to that specific clock? If not you can always take another clock source and use a PLL to generate your 200Mhz clock.

Sazzach, 2025-04-09:
>Y19/W20 are SERDES input only; you can connect them to EXTREFCLK but I don't believe there is any way to use the clock from the fabric.

I was trying to get something to work about a year ago and did some experimentation/research with respect to trying to use the EXTREFCLK with nextpnr. This is what I found:
- Diamond can use the EXTREF as a global clock
- nextpnr can't for LFE5UM5G-85
- nextpnr ***can*** use EXTREF0 as a global clock for LFE5UM5G-45
    - (EXTREF1 doesn't work, which seems strange to me)
    - I don't have a board with a LFE5UM5G-45 so I can't confirm this

Interestingly, Issue: #438 and the associated PR: #440 deal with making a SERDES example for the ECP5 VERSA board (LFE5UM5G-45) work. This example uses EXTREF0 as a clock for both the SERDES and the reset logic.

I attached two tarballs. The first contains a diamond project for the ECP5-EVN that uses EXTREF1 as a clock to blink the LEDs. It also contains a makefile that attempts to build it with yosys/nextpnr and fails with this message:
```
Info: Routing..
Info: Setting up routing queue.
Info: Routing 104 arcs.
Info:            |   (re-)routed arcs  |   delta    | remaining|       time spent     |
Info:    IterCnt |  w/ripup   wo/ripup |  w/r  wo/r |      arcs| batch(sec) total(sec)|
Warning: Failed to find a route for arc 0 of net refclko.
ERROR: Routing design failed.
3 warnings, 1 error
```

The second is a similar yosys only version of the previous project modified for the ECP5 VERSA board. It can be built via the makefile and should blink the LEDs on it. (I don't have one to test with)

[test_refclk.tar.gz](https://github.com/user-attachments/files/19660136/test_refclk.tar.gz)
[test_refclk_45.tar.gz](https://github.com/user-attachments/files/19660138/test_refclk_45.tar.gz)