YosysHQ/nextpnr #1481: nextpnr-generic fails to connect IOBs for simple wire-through

splogdes, 2025-05-12:
## Issue: 

After place & route, the IOB instances for `in` and `out` have no connections on their `PAD` or data ports, leaving the signal floating.

### Steps to Reproduce

1. Use the default configurations in `generic/examples`.
2. Modify `blinky.v` as follows:

    ```verilog
    module top(input in, output out);
         assign out = in;
    endmodule
    ```

3. Execute `simple.sh`.

### Observed Behavior

The synthesized netlist shows `in` and `out` connected to net `2` as expected:

However, the IOB primitives are misconfigured:

  ```json
  "ports": {
       "out": { "direction": "output", "bits": [ 23 ] },
       "in": { "direction": "input", "bits": [ 22 ] }
  }
  ...
  "in$iob": {
    "parameters": {
      "ENABLE_USED": "00000000000000000000000000000000",
      "OUTPUT_USED": "00000000000000000000000000000000",
      "INPUT_USED": "00000000000000000000000000000001"
    },
    "connections": { "O": [ 23 ], "EN": [  ],  "I": [  ], "PAD": [  ] }
  },
  ```
 
- `in$iob` should have `OUTPUT_USED` set to `1` and connection `"I": [ 22 ]`.

### Additional Context

Below are relevant visualizations:

- **Synthesis**:  
  [image]

- **Implementation**:  
  [image]

whitequark, 2025-05-12:
I'm not sure I see anything wrong here? `PAD` doesn't need to be connected in a post-P&R netlist since the mere act of placing an IOB at the right location enables it to control a pin. Moreover, `O` is the output of an IOB (you can tell this from the SVG because is shown on the right side of the box), which is how an input-type IOB would provide its signal to the netlist within the FPGA.

splogdes, 2025-05-13:
Ah I see sorry for that I was expecting to see some sort of connection.

Best,
Ollie

whitequark, 2025-05-13:
I guess when you talk about equivalence checking, you mean equivalence of pre- and post-legalization code, is that right? Other than legalization, nextpnr shouldn't be changing the netlist, only adding placement attributes and routing information in the JSON.

I have no authority to talk about changing nextpnr-generic (other maintainers would) but as a suggestion, would it work for you to add IOBs as a final techmapping step in Yosys? Then you could do structural equivalence checking instead of a miter, which could simplify your job.

closed 2025-05-13 by splogdes

splogdes, 2025-05-13:
@whitequark Thanks again for clarifying the PAD behaviour.

(For context: I’m working on a master’s dissertation that involves fuzz-testing P&R tools and hunting for logic-affecting mutations. Knowing that nextpnr-generic doesn’t alter netlist structure beyond placement/routing lets me narrow my search space considerably. Also the structural equivalence checking really speeds things up)

whitequark, 2025-05-13:
> Knowing that nextpnr-generic doesn’t alter netlist structure beyond placement/routing lets me narrow my search space considerably.

Placement and routing steps themselves don't affect the netlist, by definition, right? Back before I started working on these tools I wondered how people even manage to make them work reliably, but then it turned out most of it doesn't change the connectivity, which makes the checks very simple. Legalization, of course, does, and so do steps like physical optimization or resynthesis, but nextpnr as far as I know doesn't do any of those (and definitely, nextpnr-generic doesn't, simply because it would not have the necessary information due to it being generic).