YosysHQ/nextpnr #1505: Himbaechel xilinx : placer crash when handling clusters

marzoul, 2025-06-25:
I have a design targeting the board Zedboard that makes nextpnr-himbaechal crash.
The design is available at the following public link (archive tar.gz, 5MB) :
https://cloud.univ-grenoble-alpes.fr/s/tgtHJP5HaRFnXTt

Digging with gdb, I found that the crash happens within function `unbindBel()` called from placer_heap.cc:1086 when an attempt at placing a cluster failed.
https://github.com/YosysHQ/nextpnr/blob/master/common/place/placer_heap.cc#L1086

I tested with a guard added to only call the function when it won't crash :
```
if(ctx->base_bel2cell[swap.first] != nullptr)
    ctx->unbindBel(swap.first);
```

As a result, this makes nextpnr test placement for mush longer.
It says that no placement solution was found (no surprise for usage of 70% of LUTs) but at least it did not crash.

I don't have enough understanding of the little details of the algorithms and the placer implementation... the right fix may be totally different.

marzoul, 2025-06-25:
Additional details :
The crash message is the following :
```
Info: Running main analytical placer, max placement attempts per cell = 246350946.
terminate called after throwing an instance of 'nextpnr_himbaechel::assertion_failure'
  what():  Assertion failure: entry != nullptr (/usr/src/debug/nextpnr-himbaechel-git/nextpnr-himbaechel/common/kernel/base_arch.h:166)
```
And with my "guard" around the call to `unbindBel()` :
```
Info: Running main analytical placer, max placement attempts per cell = 246350946.
ERROR: Unable to find legal placement for cell 'nnawaq_inst.i_win6.gen_hwopt.mem.gen_blocks:1.mem.0.274/DPR0' of type 'SLICE_LUTX' after 246350947 attempts, check constraints and utilisation. Use `--placer-heap-cell-placement-timeout` to change the number of attempts.
46 warnings, 1 error
```

gatecat, 2025-06-25:
Thanks for the report and testcase! Hopefully the initial crash is solved more robustly by 1cd1e4a8.

I then also get the `'nnawaq_inst.i_win0.gen_hwopt.mem.gen_blocks:1.mem.0.196/DPR0'` error, I will try and see if there's anything going wrong there too but it's more likely utilization as you suspect.

You could try adding `-flatten -abc9` to the synthesis command, hopefully by reducing the number of unnecessary wide LUT/mux trees this will bring down area a bit.

marzoul, 2025-06-25:
I recompiled nextpnr just now so it has the commits mentioned by @gatecat 
The crash does not happen :-)

However the default number of attempts per cell is reduced to 10000.
I tested with the option `--placer-heap-cell-placement-timeout 246350946` just to try to mimic the original run, but this seems to have no effect...

marzoul, 2025-06-26:
Yummy ! Thank you for your contributions right to the point :-)
Indeed with #1507 it now fails to place a simple ALU that should be max 2 carry4 long (counter up to 64)... too bad.

I also tried with different random seeds, no success.
But I see that the failure always hits a carry4 from a fifo instance, just the affected fifo changes between runs.
So I am thinking that the placer may be too insensitive to random seed ? Is there a way to make nextpnr shake harder its initial placement? such as do some simulated annealing, or full discard of initial placement, etc ?

Actually the vendor tools have no issue placing and routing similar designs with much higher utilization in same chip (granted these have capability to split carry chains, but I don't expect this to be critical here). So I'll continue searching for more netlist-level pre-constructs that make it easier/possible for FOSS tools to succeed.
Related to mux implems with muxf7/8, I have a PR to clean and open for Yosys, to improve its mux mappings for xilinx technology.

marzoul, 2025-07-01:
The original crash issue is fixed so I am closing this issue.
The fact that the placer fails to find a solution is a different challenge.