YosysHQ/nextpnr #1553: OpenMP option default setting

Tectu, 2025-09-08:
While preparing the nextpnr-0.9 release for FreeBSD, I came across the CMake option `USE_OPENMP` which appears to be `OFF` by default.

What is the reason for this to be off by default?
Is this considered an "experimental" or "unstable" feature? Or could I just set it to `ON` in the FreeBSD port without any worries?

Ravenslofty, 2025-09-09:
My understanding is that enabling OpenMP is meant to help accelerate the Eigen matrix solver used in the `heap` placer. Honestly, I don't know if it actually does anything; if it does, I think there might be another win possible by enabling wider SIMD through compiler flags.

(But writing CMake to do this terrifies me.)

whitequark, 2025-09-14:
I can write the CMake for it.